ArduPilot/ardupilot #26038: OSD lost for DJI O3 Air Unit

YangChen1234, 2024-01-20:
After testing it was found，I had this problem using both the 4.4.4 and 4.4.2 firmware versions.
The following parameters are configured without OSD display for DJI O3 Air Unit:
.SERIAL4_PROTOCOL =33//DJI FPV
 SERIAL4_OPTIONS = 0 
 SERIAL4_BAUD = 115
 OSD_TYPE =3//MSP
 OSD_OPTIONS = 0
 OSD_FONT = 2

However, this configuration parameter table has OSD display  DJI O3 Air Unit
.SERIAL4_PROTOCOL = 42//DisplayPort
 SERIAL4_OPTIONS = 0 
 SERIAL4_BAUD = 115
 OSD_TYPE =5 //MSP_DisplayPort
 OSD_OPTIONS = 0
 OSD_FONT = 2

Under normal circumstances, this setting can display OSD, .SERIAL4_PROTOCOL =33//DJI FPV  ,OSD_TYPE =3//MSP
I'm not sure if this is a bug.

andyp1per, 2024-01-20:
So I think its clear that DJI O3 uses MSP DisplayPort rather than the DJI protocol. Probably needs a wiki entry @Hwurzburg

closed 2024-01-20 by andyp1per

lida2003, 2024-01-22:
> So I think its clear that DJI O3 uses MSP DisplayPort rather than the DJI protocol. Probably needs a wiki entry @Hwurzburg

Yeah, the question here is: should we use `SerialProtocol_MSP_DisplayPort` instead of `SerialProtocol_DJI_FPV` in hwdef?