ArduPilot/ardupilot #26039: ArduPilot's use of submodules and recommended workflow hose the git tree when rebasing on top of master

Ryanf55, 2024-01-20:
## Bug report

**Issue details**

When using git submodules, when they are moved around, it causes major issues when switching branches.

**Version**

* Start with HEAD at 5c1ed968b18ea196b8376cdd6cc7aeae9b3085c4 then try rebase on f67d8956778de0ed1e5ddebf159ecd1ca0f6f3d7

* git version: 2.34.1

**Platform**
[x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Steps to reproduce**

1. Start with an empty directory
1. Clone my repo as origin ryanf55/ardupilot with --recurse-submodules at hash 5c1ed968b18ea196b8376cdd6cc7aeae9b3085c4
1. Change directory into ardupilot
1. Add an "upstream" remote of ardupilot/ardupilot
1. Fetch upstream
1. Rebase origin on upstream
1. Congrats, the work tree is hosed

**Logs**

Full logs to reproduce this are attached.
[log.txt](https://github.com/ArduPilot/ardupilot/files/13999273/log.txt)

The exact error occurs here:
```
ryan@B650-970:~/Dev/tmp_ws/ardupilot$ git rebase upstream/master
fatal: not a git repository: ../../.git/modules/modules/lwip
fatal: could not reset submodule index
error: could not detach HEAD
```

**Recommended Process**

1. Add documentation to the wiki on how to switch branches and rebase without error
OR
1. Switch to a different version control tool for dependencies such as Conan, CPM, VCS, Google Repo

tridge, 2024-01-21:
I just followed your list of steps and had no issues. All steps worked fine

tridge, 2024-01-21:
@Ryanf55 I've now reproduced. The key step which you didn't call out in the description is "git config submodule.recurse true". If you don't do that then it all works fine.

Ryanf55, 2024-01-21:
> @Ryanf55 I've now reproduced. The key step which you didn't call out in the description is "git config submodule.recurse true". If you don't do that then it all works fine. I've never set that config option myself, and just use --recurse when I want that (I often don't)

Thanks for looking at this and reproducing the issue!

I'll create a Wiki PR to recommend turning off the `submodule.recurse` locally in the ArduPilot repo. ^

peterbarker, 2025-05-11:
@Ryanf55 did you create that Wiki note?

Ryanf55, 2025-05-11:
We found it was a bug in git, and is only seen with nondefault settings.

closed 2025-05-11 by Ryanf55