ArduPilot/ardupilot #26050: Bootloader data writing protocol needs to be improved

lixiaowei123, 2024-01-22:
The protocol for writing data to bootloader needs to be improved. By default, it is verified after writing. I think this is very unsafe and requires verification every frame.

[image]

peterbarker, 2024-07-31:
I don't know how checking after every write helps at all.

Your bootloader is hosed after a single bad write - whether we detect that straight-away or by doing the checksum isn't important.

Given you also need to erase entire sectors to fix the bad write, even working out that a particular page write failed isn't helpful.

Perhaps I'm misunderstanding the problem you're outlining.

peterbarker, 2024-09-04:
Closing this as NeedAnswer for several weeks.

@lixiaowei123 if you think I'm wrong, please reopen!

closed 2024-09-04 by peterbarker