ArduPilot/ardupilot #26053: Yaapu receiving out of order/missing mavlink messages through fport telemetry.

GalacticGreg, 2024-01-22:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

I have the newest yaapu release (Mar 2023) for my taranis X7 accst with an R-XSR in fport mode (using inverted solder pad) on an omnibusf4. Mavlink messages are corrupted. I tried every permutation of pull-up on the TX pin, fport pad option and both 57600 and 115200 baud rates so far and the corruption remains.

**Version**
4.5dev, edgetx 2.9.2, RXSR fport solder pad v. 2.1.1 firmware, yaapu 2.0

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X4

**Hardware type**
Custom hardware definition omnibusf4.

**Logs**
_Please provide a link to any relevant logs that show the issue_

GalacticGreg, 2024-02-04:
Just a note to anyone in the same situation, downgrading to R-XSR F.Port firmware version 2.1.0 from 2.1.1 fixes this issue. I believe 2.1.1 was supposed to fix no output from the receiver during a failsafe, but I can confirm that when failsafe is set to "no pulses" then arducopter detects the failsafe properly.

mpjf01, 2024-02-21:
#25546 might be related

GalacticGreg, 2024-02-21:
I don't think it is mpjf01. In #25546 the issue was with ardupilot only and with the newer ACCESS firmware.

My problem is with the older ACCST firmware and even Betaflight users are having problems with telemetry in 2.1.1 firmware as per reports from 3 years ago. Frsky was releasing beta versions of the firmware to address this issue at the time then gave up, closed their beta repository and ghosted anyone complaining about ACCST firmware problems to focus on the newer ACCESS, leaving behind bugs they introduced while trying to fix other problems.

This issue is not worth addressing through ardupilot, I'm going to close it. At least it is documented for anyone else stuck with old hardware like me and wondering what's going on.

closed 2024-02-21 by GalacticGreg