ArduPilot/ardupilot #26070: Request to remove or alter auto config for Septentrio GNSS receiver

SeptentrioGNSS, 2024-01-23:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request
In the current and latest stable version of the ArduPilot code for the implementation of Septentrio receivers (SBF), there is an auto config for a logged stream to the internal disk of the receiver. The mention of this stream is in the file: **AP_GPS_SBF.h**. Is there a possibility to either remove this or make it adaptable for the user to specify if they want this stream logged. 

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
Currently a setSBFOutput command is sent to the receiver. It overwrites configurations made on the receiver side or it might not be needed by the user to have this log and there is no way of removing this from the auto config.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
In an ideal world we adapt the current system to have three user defined options.
1) Having the option to not send the command at all. All the configuration (if needed) of the logging of data can be done on the receiver side and is not touched by the autopilot software. 
2) A secondary option where the (generic) Support block is included to the current receiver configuration. This is beneficiary for post processing the data as Septentrio commonly uses this at the start of troubleshooting the GNSS receiver. It allows user to have a 
It would look like: "sso,Stream2,Dsk1,**+Support+PostProcess**,msec100"
3) Finally a more advanced option could be that the SD card logging config happens on the autopilot side and options for selecting what blocks are needed to log on the receiver. As this involves complex code probably, the first two options would be sufficient to start.

On the other side, more control on the interval could be helpful. The current setup has a fixed 100msec interval, it could become a parameter in Mission Planner.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Check above to implement 1) as this allows user to configure receiver side if they need it. 

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._
[image]
code in question is at line 87

peterbarker, 2024-02-12:
Ping @WickedShell , who is responsible for the current logging settings and might have a vested interest in the current settings

WickedShell, 2024-02-15:
There are a couple of things we can do here, listed in ease of things that already exist:

- You can add custom configuration lines to the hwdef and do a custom build, which would allow specifying a different logging stream to be specified. This will technically send the hard coded line first, then follow it up with the custom one, which isn't great, but would work. The other downside is that this requires you to have a custom build which is less commonly available.
- You can send gps data injections from the GCS, there's exsisting MAVLink handling for this, but I don't think there is a convenient interface in MP for it. I've used this in a custom GCS to inject 1 way configuration messages to the GPS driver. This is annoying to do every boot though, and obviously is pretty custom again.
- Now we are at the spot of new changes. The easiest change is to make it so that if GPS_RAW_DATA was set to 0 to not emit the logging message at all from the configuration. Adding more meanings to this is a bit harder, as we've already made 2 have a special case for Septentrio GPS units (unmount the SD card when disarmed).
- Do something more complicated, as proposed above, but which is a bit harder to generalize for all units, although we could lean against the per driver specific parameters.

Regarding the comments made, I'd really prefer the ArduPilot implementation of logging continue to request `Comment+ReceiverStatus` as from a support perspective I've found both of these to be extremely useful, and pretty light weight. And if you are logging `PostProcess` having `Event` seems much more useful, and again if it's not connected it should be pretty low impact. Are you seeing problems with this?

My builds have been adding the `Support` block as I've found it to be valuable and nice. (I can be convinced it should be in the default logging set). If we add that in, that would only leave us needing to add a flag for `BBSamples` and rates I guess.

flyingthingsintothings, 2024-03-01:
I would like to work on this issue. I discussed the implementation with @SeptentrioGNSS and we had the following idea for a more flexible logging implementation.

There could be three new parameters added in the `GPS` module. The first would allow the user to choose a logging frequency from a set of discrete values (Off (0Hz), 0.1Hz, 0.5Hz, 1Hz... 20Hz). This frequency would then be used on a best-effort basis by every driver that wants to support custom logging frequency. Users could choose Off (0Hz) to disable logging completely which would prevent any logging setup from being performed on the receiver. The second parameter would represent a logging level that is then translated by each driver into actual data that makes sense for that level (e.g. `Lite` logs only contain `Comment+ReceiverStatus` on Septentrio GNSS receivers, `Basic` contains `Lite` + `PostProcess+Support`...). The levels could be called `Lite`, `Basic`, `Default` and `Full`. The specific data at each level for Septentrio GNSS receivers isn't decided yet, but would be properly documented so users know which level logs what. The third parameter would be used to indicate whether logging should be added to any existing logging that is already on the receiver, or would overwrite it.

Suggested names for the parameters would be `GPS_LOG_HZ` for the logging frequency, `GPS_LOG_LEVEL` for the logging level and `GPS_LOG_FORCE` for log overwriting.

The advantage of this approach is that the parameters are detailed enough for users to configure logging how they want, while being generic enough to support different drivers/receivers without adding driver-specific parameters. Any driver-specific things could be documented so users are aware whether their receiver will support the chosen parameter values.

I would love any input on this approach.
@WickedShell

peterbarker, 2024-03-07:
@flyingthingsintothings  created a PR here: https://github.com/ArduPilot/ardupilot/pull/26442