ArduPilot/ardupilot #26077: Plane: Consecutive Turn Support

jrbronkar, 2024-01-25:
## Feature request
### Background
I have already worked on the merged https://github.com/ArduPilot/ardupilot/pull/25873 to unblock this work - I will be discussing from both a developer perspective.
### Summary of Desired Feature
I want to fly this relatively simple curve (flying left to right) that is kinematically possible to fly on a normal Plane and which consists of a straight (starting at blue) into a quarter circle, then into another quarter circle, then into a straight (exit at green). I believe this is impossible using ArduPilot primitives (LOITER_TURNS, Waypoints) in a Mission on ArduPlane as it stands today - and in fact all precise consecutive turns are impossible, see "Proposed Developer Diagnosis of Issue".
[image]
### Impossibility Explanation (using Primitives)
To fly this path, you might try to upload something like the below with ArduPilot Primitives.
[image]
Flying this path from left->right:
1. Waypoint to specify entrance to turn
2. Clockwise LOITER_TURN with desired radius + center to represent first quarter turn.
3. Waypoint to indicate exit of first LOITER_TURN (denoted by middle hollow point)
4. CounterClockwise LOITER_TURN with desired radius + center to represent second quarter turn.
5. Final waypoint somewhere along the green segment so you know when to exit the second LOITER_TURN.

 But it doesn't seem to work - why? (note: above proposed path maps to waypoints 3-7)
[image]
1. We reach Waypoint 3
2. Switch onto first LOITER_TURN and ArduPilot checks exit criteria
a. Exit criteria is `verify_loiter_heading` for [both branches of this function!](https://github.com/ArduPilot/ardupilot/blob/8c72304ab912b42d38bd3301fd964aaec7149e77/ArduPlane/commands_logic.cpp#L759-L767) of the `verify_loiter_turns` function (Note: this means LOITER_TURNS has [heading required](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_TURNS) in all cases!)
b. verify_loiter_heading [calls isHeadingLinedUp](https://github.com/ArduPilot/ardupilot/blob/8c72304ab912b42d38bd3301fd964aaec7149e77/ArduPlane/commands_logic.cpp#L1175) with the next location
c. isHeadingLinedUp [hits logic that breaks out of the loiter immediately](https://github.com/ArduPilot/ardupilot/blob/8c72304ab912b42d38bd3301fd964aaec7149e77/ArduPlane/mode_loiter.cpp#L45-L51) - since the "next" waypoint, or Item 5 is within the loiter radius + 5% (it is on the loiter radius)
3. Now that we have broken out of the first LOITER_TURN we transit normally to Item 5, but we are nearly perpendicular to the desired second loiter.
4. We reach item 5, and we switch quickly onto the second LOITER_TURN and we check exit criteria - isHeadingLinedUp again - and we point towards the next and final waypoint (item 7) pretty immediately and exit the second LOITER_TURN

Great. Shouldn’t place waypoints  after a loiter within the loiter radius + 5%, got it. What if I just remove that waypoint then?
[image]

Doesn't work either! we
1. Reach Waypoint 3, switch onto first LOITER_TURN
2. Stay on first LOITER_TURN for a very small time after waypoint 3 until `isHeadingLinedUp` with the center(!) of the second LOITER_TURN.
3. Switch to the second LOITER_TURN. As we transit to the second LOITER_TURN, we reach isHeadingLinedUp with waypoint 6 and exit the second LOITER_TURN loop (even though we haven't reached the loiter yet) and start transiting to waypoint 6.

### Proposed Developer Diagnosis of Issue
isHeadingLinedUp is required in all cases for a LOITER_TURN, and is relatively inflexible - it does not allow for waypoints to be on the radius of the circle + 5%, that one otherwise might be able to use as exit criteria for a discrete turn. 

Even if you operate under the world of [fractional turns](https://github.com/ArduPilot/ardupilot/pull/25873) that recently merged, you wouldn't be able to chain two quarter LOITER_TURNS together to implement the desired smooth consecutive turn path, because the isHeadingLinedUp check is required in all cases. you would finish a quarter of the turn, and then try to turn to get heading with the center of the next turn, then go back to  the second turn, then get heading with the next point, instead of: finish quarter of the first turn, switch to second turn cleanly, finish quarter of second turn, continue on path. I can demonstrate this in the below comments if anybody would like to see in SITL.

**Describe the solution you'd like**
I would propose that both elements of this should be fixed, though I would note either should resolve the issue, albeit at different levels:
> isHeadingLinedUp is required in all cases for a LOITER_TURN
> isHeadingLinedUp is relatively inflexible

1. Implement Heading Required field for [ArduPlane LOITER_TURNs](https://ardupilot.org/plane/docs/common-mavlink-mission-command-messages-mav_cmd.html#mav-cmd-nav-loiter-turns) to align with the [mavlink spec](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_TURNS) 
- If we implemented the Heading Required field, you would be able to use fractional turns + heading required disabled/enabled checks to correctly chain together turns.
2. Make the isHeadingLinedUp more flexible. See https://github.com/ArduPilot/ardupilot/pull/25938 as a proposed solution to more dynamically configure the exit criteria based where that "next" waypoint is. Currently the heading tolerance used for the exit criteria is hard coded. https://github.com/ArduPilot/ardupilot/blob/8c72304ab912b42d38bd3301fd964aaec7149e77/ArduPlane/mode_loiter.cpp#L61-L82
- If you implemented this, you could fly the first example cleanly without running into breakout logic.
- See [PR Desc](https://github.com/ArduPilot/ardupilot/pull/25938) for a nice SITL demonstration!: 
[image]

**Describe alternatives you've considered**
See description of two options above, I consider both of these to be valid approaches that should both land on master. I would think landing the already implemented [patch PR](https://github.com/ArduPilot/ardupilot/pull/25938) before the likely more intensive "implement heading required" makes sense.

Phew. That was long. I have given this a deal of thought, and I would appreciate any thoughts, points of discussion, or other ideas for solutions.
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

timtuxworth, 2024-01-25:
Here's another option - implement it in Lua by basically writing this as an "aerobatic trick". Checkout the "Scripted Aerobatics" examples https://ardupilot.org/plane/docs/common-scripting-aerobatics.html.

IamPete1, 2024-01-25:
Do you actually want to program such moves with loiter turns, or if we were to do it automatically would what be better? It would be a big re-work, but we could update to use Dubins paths for FW navigation.

jrbronkar, 2024-01-27:
I'd like to program such moves with loiter turns. I am willing to work on it as I did for the fractional loiter turn support, and the changes should be similar net benefits to Missions by expanding the use cases of loiter turns to either A) align more fully with the mavlink spec, B) be more flexible and less hardcoded ([heading check patch](https://github.com/ArduPilot/ardupilot/pull/25938)), or C) both (A) and (B). 

My desire for this issue was detail out this problem I have been noticing in SITL and seeing if my proposed fixes seem sane and reasonable to the problem. perhaps we can chat at the next dev call?

IamPete1, 2024-01-28:
One problem I found while trying this out is that because we scale the loiter radius with altitude its very hard to manually program such missions if your above sea level because the GCS's shown radius's are not the what we will fly.

This also highlights a bug in the original break out code because it assumes we will fly at the un-corrected radius. I wonder if this is why we needed those fudge factors and increasing error bound in the first place....

In any case I have done a alternate fix here: https://github.com/IamPete1/ardupilot/tree/loiter_heading_linedup

jrbronkar, 2024-01-29:
gotcha. appreciate that flag. I will take a look at your branch, thank you!

jrbronkar, 2024-01-31:
This [PR](https://github.com/ArduPilot/ardupilot/pull/26102) appears to resolve this issue (represents the branch linked above that Pete suggested).  tagging for visibility into the smooth turn issues this was geared, in part, at addressing.

magicrub, 2024-02-01:
For what it's worth, I disable that altitude radius thing because we've implemented it wrong. Its meant to keep us in our flight envelope but it should be enforcing a minimum, not always scaling. Not being able to actually control your aircraft path is kind of a party foul.

IamPete1, 2024-02-01:
> For what it's worth, I disable that altitude radius thing because we've implemented it wrong. Its meant to keep us in our flight envelope but it should be enforcing a minimum, not always scaling. Not being able to actually control your aircraft path is kind of a party foul.

I agree.

Ryanf55, 2024-02-03:
I'm super interested this, or a more generic capability for a trajectory-following controller following 3D dubins curves.

This could be used in #2380 , and allow ArduPilot to allow ROS to generate more intelligent paths than what was shown possible here: https://github.com/ArduPilot/ardupilot/pull/25722

Would you all consider just implementing a trajectory-following controller?

jrbronkar, 2024-02-06:
Chatted at the most recent dev call - going to put some notes here for reference.

Sounds like there is some excitement about potential future dubins' planning/path support for ArduPlane but it also sounds like we can move ahead with https://github.com/ArduPilot/ardupilot/pull/26102 to unblock this issue in the absence of more complex fixes short-term as it is a net benefit in improving use/usage of the current primitives.

jrbronkar, 2024-04-07:
closing. after extensive testing: https://github.com/ArduPilot/ardupilot/pull/26102 has resolved the issue.

closed 2024-04-07 by jrbronkar