ArduPilot/ardupilot #26083: accel to lean angles should be modified

lixiaowei123, 2024-01-26:
I recently found that the handling of acceleration tilt angle in Ardupilot is not detailed enough, and the annotations are not perfect enough:
`void AC_PosControl::accel_to_lean_angles(float accel_x_cmss, float accel_y_cmss, float& roll_target, float& pitch_target) const
{
    // rotate accelerations into body forward-right frame
    const float accel_forward = accel_x_cmss * _ahrs.cos_yaw() + accel_y_cmss * _ahrs.sin_yaw();
    const float accel_right = -accel_x_cmss * _ahrs.sin_yaw() + accel_y_cmss * _ahrs.cos_yaw();

    // update angle targets that will be passed to stabilize controller
    pitch_target = accel_to_angle(-accel_forward * 0.01) * 100;
    float cos_pitch_target = cosf(pitch_target * M_PI / 18000.0f);
    roll_target = accel_to_angle((accel_right * cos_pitch_target)*0.01) * 100;
}`

theoretical basis:
[image]
[image]
however:az!=0
[image]

modified:
[image]

``void AC_PosControl::accel_to_lean_angles(float accel_x_cmss, float accel_y_cmss, float& roll_target, float& pitch_target) const
{
    //
     float az=_ahrs.get_accel_ef_blended().z;
    //Todo: Align target yaw from NED to forward-right-down ,not a body frame. however rotate accelerations into body forward-right frame
    const float accel_forward = accel_x_cmss * _ahrs.cos_yaw() + accel_y_cmss * _ahrs.sin_yaw();
    const float accel_right = -accel_x_cmss * _ahrs.sin_yaw() + accel_y_cmss * _ahrs.cos_yaw();

    // update angle targets that will be passed to stabilize controller
    //pitch_target = accel_to_angle(-accel_forward * 0.01) * 100;
    //float cos_pitch_target = cosf(pitch_target * M_PI / 18000.0f);
    //roll_target = accel_to_angle((accel_right * cos_pitch_target)*0.01) * 100;
    pitch_target = atanf(-accel_forward / ((GRAVITY_MSS-az) * 100.0f)) * (18000.0f / M_PI);
    float cos_pitch_target = cosf(pitch_target * M_PI / 18000.0f);
    roll_target = atanf(accel_right * cos_pitch_target / ((GRAVITY_MSS-az) * 100.0f)) * (18000.0f / M_PI);
}`

amilcarlucas, 2024-01-26:
You need to use three ``` symbols for quotes to work. But a PR is even better

lthall, 2024-01-26:
This issue correctly points out that we do not include the vertical acceleration into our lean angle calculation. Instead we assume the vertical acceleration is always zero and the aircraft is always experiencing 1 g.

We do not include the vertical acceleration into this part of the controller because it would provide a noise and feedback path that will compromise performance and complicate analysis. Including vertical acceleration does not promise to improve performance in normal flight because the altitude controller keeps the vertical acceleration at near zero the majority of the time.

I have considered a similar approach where I include the commanded vertical acceleration to provide a noise free correction to the lean angle calculation during vertical speed changes. However the times when this will have any impact will be short and do not justify the cost in complexity.

I have also considered including feedback based on the current trust level to potentially alleviate an oscillatory condition with slung payloads. In this case the vertical acceleration is still zero but the effective mass of the aircraft is changing. The suggestion above would not address this problem.

I am going to close this issue for the reasons above but I appreciate your review of the code and your effort in explaining your suggestion.

closed 2024-01-26 by lthall