ArduPilot/ardupilot #26088: Add documentation for undocumented LogMessages

shancock884, 2024-01-27:
## Feature request

According to the list dumped out by the Autotest log, there are a number of log messages without documentation.
It would be nice to add details of them.
It looks like there was a push a couple of years ago to clear the list at the time (ref #13900), so I guess a lot of these are new since then.

**Is your feature request related to a problem? Please describe.**
An improvement request, rather than a problem, but certainly details of the more common ones would help users.

**Describe the solution you'd like**
Crowd-sourced documentation

**Describe alternatives you've considered**

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
This is the list, including the file in which the message is mentioned, so people can see which module it lives in:

ArduPlane/quadplane.cpp:    "FWDT",
ArduPlane/quadplane.cpp:    "QPOS"

AC_Autorotation/AC_Autorotation.cpp:    "AROT",
AC_AutoTune/AC_AutoTune_Heli.cpp:        "ATDH",
AC_AutoTune/AC_AutoTune_Heli.cpp:        "ATNH",
AC_AutoTune/AC_AutoTune_Heli.cpp:        "ATSH",
AC_CustomControl/AC_CustomControl.cpp:    "CC"
AP_AHRS/AP_AHRS_DCM.cpp:        "DCM"
AP_DroneCAN/AP_DroneCAN.cpp:        "CVOL",
AP_DroneCAN/AP_DroneCAN.cpp:    "CANS",
AP_Generator/AP_Generator_IE_2400.cpp:                "IE24",
AP_Generator/AP_Generator_IE_2400.cpp:                "IEFC",
AP_Generator/AP_Generator_RichenPower.cpp:        "GEN",
AP_GPS/LogStructure_SBP.h:      "SBRE"
AP_GPS/LogStructure_SBP.h:      "SBRM"
AP_GyroFFT/AP_GyroFFT.cpp:        "FTN3",
AP_InertialSensor/AP_InertialSensor_Backend.cpp:    "IREG"
AP_InertialSensor/AP_InertialSensor_tempcal.cpp:    "TCLR"
AP_InertialSensor/LogStructure.h:      "ISBD"
AP_InertialSensor/LogStructure.h:      "ISBH"
AP_Logger/LogStructure.h:      "FILE"
AP_Logger/LogStructure.h:      "VER"
AP_Mount/SoloGimbal.cpp:        "GMB1",
AP_Mount/SoloGimbal.cpp:        "GMB2",
AP_Relay/AP_Relay.cpp:        "RELY"
AP_Soaring/AP_Soaring.cpp:    "SORC"
AP_TECS/AP_TECS.cpp:            "TEC3"
AP_TemperatureSensor/AP_TemperatureSensor_Backend.cpp:    "TEMP",

SITL/SIM_Blimp.cpp:  "SAF1"
SITL/SIM_Blimp.cpp:  "SAN1"
SITL/SIM_Blimp.cpp:  "SAN2"
SITL/SIM_Blimp.cpp:  "SBA1"
SITL/SIM_Blimp.cpp:  "SBLM"
SITL/SIM_Blimp.cpp:  "SFA1"
SITL/SIM_Blimp.cpp:  "SFAN"
SITL/SIM_Blimp.cpp:  "SFN"
SITL/SIM_Blimp.cpp:  "SFT"
SITL/SIM_Blimp.cpp:  "SFV1"
SITL/SIM_Blimp.cpp:  "SMGC"
SITL/SIM_Blimp.cpp:  "SRT1"
SITL/SIM_Blimp.cpp:  "SRT2"
SITL/SIM_Blimp.cpp:  "SRT3"
SITL/SIM_Blimp.cpp:  "SSAN"

To get the list I downloaded the full log from autotest website, and searched for "Undocumented message".

IamPete1, 2024-01-27:
We really should lock this in with a auto-test once we get them all documented again. Or maybe even have white list for logs that are "allowed" to be undocumented so we can at least ensure were not adding and new ones without docs.

peterbarker, 2025-05-11:
> We really should lock this in with a auto-test once we get them all documented again. Or maybe even have white list for logs that are "allowed" to be undocumented so we can at least ensure were not adding and new ones without docs.

Done: https://github.com/ArduPilot/ardupilot/pull/30035

closed 2025-10-01 by shancock884