ArduPilot/ardupilot #26094: THROTTLE_NUDGE doesn't work in GUIDED Mode

AutomationEngineer, 2024-01-29:
Having parameter THROTTLE_NUDGE set to 1, throttle stick have no effect on airspeed target value when in GUIDED mode while have an expected effect in AUTO mode.

Plane 4.3.8

Quadplane

CUAV X7 Pro Plus

The issue is probably dew to use of `is_zero` in navigation.cpp:
```
#if OFFBOARD_GUIDED == ENABLED
    if (control_mode == &mode_guided && !is_zero(guided_state.target_airspeed_cm) && (airspeed_nudge_cm != 0)) {
        airspeed_nudge_cm = 0; //airspeed_nudge_cm forced to zero
    }
#endif
```
which is not correct in this case, cause `guided_state.target_airspeed_cm` initiates to -1
other parts of code uses comparison to 0.0 instead, which is correct.

IamPete1, 2024-01-29:
A PR to fix that would be great. We should bring them all in line, I see 3 places its used, here we have `!is_zero`, `> 0` and for logging `is_positive`. They should probably all be `is_positive`.

AutomationEngineer, 2024-01-31:
I've added a PR #26107, but it didn't pass plane SITL tests, however have passed quadplane tests.
Loking at logs showed some issues with fence handling (I can't imagine how it is related) and an issue with change altitude command in GUIDED mode, which is directly related to particular changes of code.
Plane didn't climb from 35 to 49 meters at some time, but could this be dew not neutral throttle, and low climb rate for this reason?
Could please somebody review the PR, my SITL logs and testing logs to point me where to move forward?

AutomationEngineer, 2024-03-29:
As PR is now passing the tests could we include it to Plane 4.5.x or should I rework it to make more "clean" (no loose comits)?

peterbarker, 2025-05-11:
@AutomationEngineer did you mean to close the associated PR?

AutomationEngineer, 2025-05-12:
It was somehow closed automatically with no changes because of my clumsy action on my fork. Not enough experience and have no one to guide me through to accomplish this as intended, sorry..