ArduPilot/ardupilot #26137: Allow using virtual environment on Ubuntu Jammy when running install-prereqs-ubuntu.sh

Ryanf55, 2024-02-04:
## Feature request

**Is your feature request related to a problem? Please describe.**

ArduPilot python packages that are installed in user-space can interfere with other toolchains. Python developers recommend virtual environments whenever possible. I find myself having to wipe user python packages because they have version incompatibilities, and other development toolchains assume python packages available outside a virtual environment are all managed properly by the distro. 

**Describe the solution you'd like**

Allow `install-prereqs-ubuntu.sh` to have the option to install everything in a virtual environment on Ubuntu Jammy.

**Describe alternatives you've considered**

Manually installing packages in a virtual environment.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

N/A

Ryanf55, 2024-02-06:
Dev call said this would be accepted if it did not impact the existing flow. An additional param, with a wiki change, would be accepted.

P81000, 2024-02-07:
I'm using a Docker container to build the code using an Ubuntu image. I was able to run `install-prereqs` inside the container and everything worked well. Can you provide more context? I'm a new ardupilot developer.

Ryanf55, 2024-02-07:
Prototype branch: https://github.com/Ryanf55/ardupilot/tree/venv-option-on-jammy

Ryanf55, 2024-09-07:
@peterbarker Would you have any time to consider adding this? The recent PR you introduced to install a different version of setuptools breaks Ubuntu jammy ROS users; now `colcon test` won't find any python tests.
 See here for context.
https://github.com/ArduPilot/ardupilot/issues/27925#issuecomment-2336227563

peterbarker, 2025-05-11:
I think I saw a PR for jammy but I can't find it now.  I think we'd allow it as an option you could turn on on the command-line or as an environment variable or something?

@Ryanf55 am I mis-remembering such a PR?

Ryanf55, 2025-05-11:
Yea, I had a branch, but couldn't finish it because the script was more complicated than I expected and I ran out of time. 
See this comment for the WIP branch: https://github.com/ArduPilot/ardupilot/issues/26137#issuecomment-1932619322