ArduPilot/ardupilot #26227: Command_Int UNSUPPORTED

nraines2002, 2024-02-14:
## Bug report

**Issue details**

Sending MAVLink messages via Command_Int is returning COMMAND_ACK:[COMMAND_NAME]:UNSUPPORTED. I am using MAVROS built using ROS2 Foxy to pack and send these commands, using the Command_Long MAVROS service to send a Set_Mode message works but returns COMMAND_ACK:DO_SET_MODE:DENIED. I have sent the message with frames, MAV_FRAME_GLOBAL, MAV_FRAME_LOCAL_NED, MAV_FRAME_MISSION, MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_FRAME_GLOBAL_RELATIVE_ALT_INT to no avail.

Current usage:

`auto request = std::make_shared<mavros_msgs::srv::CommandInt::Request>();`
`                request->frame = mavros_msgs::msg::Waypoint::FRAME_GLOBAL_REL_ALT;`
`                request->command = 113;`
`                request->current = 0;`
`                request->autocontinue = 0;`
`                request->param1 = 0;`                        
`                request->param2 = 0;`                        
`                request->param3 = 0;`                        
`                request->param4 = 0;`                        
`                request->x = 0;`                        
`                request->y = 0;`                        
`                request->z = alt; (altitude in meters)`
`                                                                      `
`                auto result = comm_client_->async_send_request(request);`

**Version**
Copter-4.4/4.3/4.2

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

**Hardware type**
None, used SITL

**Logs**

Current usage:

rmackay9, 2024-02-15:
@nraines2002,

Which version of AP is on the autopilot?  4.5.0 is required for many commands when using command_int.

amilcarlucas, 2024-02-15:
This is a support question that I already answered on the forum.

nraines2002, 2024-02-15:
> @nraines2002,
> 
> Which version of AP is on the autopilot? 4.5.0 is required for many commands when using command_int.

I was trying various different versions and just tried 4.5.0 with no luck.

[image]

amilcarlucas, 2024-02-15:
Then that command is not yet supported.You should change the title of this issue to:
CONDITION_CHANGE_ALT COMMAND_INT unsupported

rmackay9, 2024-02-15:
OK, good that we're getting to the real issue now.

@nraines2002, have you seen [this wiki page that documents how copters can be controlled using MAVLink](https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html)?

I'm not too enthusiastic about adding support for [CONDITION_CHANGE_ALT](https://mavlink.io/en/messages/common.html#MAV_CMD_CONDITION_CHANGE_ALT) when we already have so many more capable alternatives.

nraines2002, 2024-02-16:
@rmackay9 I haven't seen that wiki, that's exactly what I needed.

The MAV_CMD_DO_REPOSITION command is all I need, thank you.

rmackay9, 2024-02-16:
@nraines2002,

great stuff.  I'll go ahead and close this issue then.  Txs for using AP!

closed 2024-02-16 by rmackay9