ArduPilot/ardupilot #26271: Unexplained spikes in EKF Position innovations

EosBandi, 2024-02-20:
Copter 4.4.4, OrangeCubePlus, Dual Here4 gps.
Logs show spikes in EKF Position innovations, some are close to the error threshold.
IMU0/Lane1 and IMU1/Lane2 shows frequent spikes, but IMU2/Lane3 show only a small amount, occasional and smaller spikes.
No sensor affinity is set, all lanes are using the same blended GPS and first compass.
Compass checked, calibrated and in sync with predicted magfield values.
Magnitude of the IPN and IPE spikes are related to the actual groundspeed.
No spikes on any other innovation values.
Log files : https://1drv.ms/f/s!AmcHEcCG2Jqso6pDjcnW3KHbxATXgQ?e=LPDX6s

amilcarlucas, 2024-02-20:
Nice and detailed bug report

tridge, 2024-02-21:
no replay on log unfortunately
the spikes are clearer in the XKF3.IPx data (which have units of meters)
[image]

tridge, 2024-02-21:
@EosBandi are you able to get a log with LOG_REPLAY=1 and LOG_DISARMED=3 ?

tridge, 2024-02-21:
I suspect the issue may be some high frequency noise in the IMU data that doesn't show up in our filtered IMU logging, but would show up with replay (EKF uses unfiltered IMU data)
it could be timing in the GPS data too, though I can't see it in this log

EosBandi, 2024-02-22:
> @EosBandi are you able to get a log with LOG_REPLAY=1 and LOG_DISARMED=3 ?

Sure, but it will take some time; the vehicles are in Bogota, the pilot is in Spain, and I'm in Hungary. I'll post the logs as soon as I manage to get the pilot and the vehicles to one place :)

bugobliterator, 2024-02-23:
It seems like an issue specific to this vehicle. I did flight with Here4 and these spikes are not present. The flight was not with blended GPS though and with moving baseline enabled.

[image]

Full Log: https://droneshare.cubepilot.com/#/s/log?b=1707286594399&c=71651630-c580-11ee-8113-bfa0943c0b8f&d=GPS+Yaw+Acceptance+Gate+Test2&e=2&a=cbfcd2fc-3fa9-11ed-b631-b97a8244eade

EosBandi, 2024-02-23:
> It seems like an issue specific to this vehicle. I did flight with Here4 and these spikes are not present. The flight was not with blended GPS though and with moving baseline enabled.
> 
To me it seems to have the same random spikes, just a little smaller magnitudes.... 
[image]

Am I just overreacting to something normal with EKF3 ?