ArduPilot/ardupilot #26281: Compilling error in Simulation environment

ApoloM, 2024-02-21:
https://github.com/ArduPilot/ardupilot/blob/e01e697343cf3eb9b0991d7153f135b60b96f965/libraries/SITL/SIM_ADSB.cpp#L86

Hello guys, I was compilling the package in my machine and found the following error:
[image]

Then, I realized that we can have different types for position and velocity vectors, since we could define or not HAL_WITH_POSTYPE_DOUBLE and HAL_WITH_EKF_DOUBLE.
In my case position was float while velocity was double.

Isn't that a bug?
Or am I missing something in my setup?
Thanks, for your help.

IamPete1, 2024-02-21:
We need some more info. Which branch? Which target? Which compiler? You can post the output of the configure step to get compiler version.

IamPete1, 2024-02-21:
I get the same error here.

I suspect you actually want:
`../Tools/autotest/sim_vehicle.py --map --console -f quadplane`

`--p` is forcing to single precision. `--postype-single    force single precision postype_t`

We might still want to fix it tho, @peterbarker Would we ever want single precision SITL with SIM on hardware?

ApoloM, 2024-02-21:
Thanks, for the response. It solved my problem.

closed 2024-02-21 by ApoloM