ArduPilot/ardupilot #26285: New attitude error calculation

lixiaowei123, 2024-02-22:
I think the attitude error calculation of APM lacks theoretical support and is not as good as the latest attitude error calculation of PX4
1. PX4 Code:
matrix::Vector3f AttitudeControl::update(const Quatf &q) const
{
	Quatf qd = _attitude_setpoint_q;

	// calculate reduced desired attitude neglecting vehicle's yaw to prioritize roll and pitch
	const Vector3f e_z = q.dcm_z();
	const Vector3f e_z_d = qd.dcm_z();
	Quatf qd_red(e_z, e_z_d);

	if (fabsf(qd_red(1)) > (1.f - 1e-5f) || fabsf(qd_red(2)) > (1.f - 1e-5f)) {
		// In the infinitesimal corner case where the vehicle and thrust have the completely opposite direction,
		// full attitude control anyways generates no yaw input and directly takes the combination of
		// roll and pitch leading to the correct desired yaw. Ignoring this case would still be totally safe and stable.
		qd_red = qd;

	} else {
		// transform rotation from current to desired thrust vector into a world frame reduced desired attitude
		qd_red *= q;
	}

	// mix full and reduced desired attitude
	Quatf q_mix = qd_red.inversed() * qd;
	q_mix.canonicalize();
	// catch numerical problems with the domain of acosf and asinf
	q_mix(0) = math::constrain(q_mix(0), -1.f, 1.f);
	q_mix(3) = math::constrain(q_mix(3), -1.f, 1.f);
	qd = qd_red * Quatf(cosf(_yaw_w * acosf(q_mix(0))), 0, 0, sinf(_yaw_w * asinf(q_mix(3))));

	// quaternion attitude control law, qe is rotation from q to qd
	const Quatf qe = q.inversed() * qd;

	// using sin(alpha/2) scaled rotation axis as attitude error (see quaternion definition by axis angle)
	// also taking care of the antipodal unit quaternion ambiguity
	const Vector3f eq = 2.f * qe.canonical().imag();

	// calculate angular rates setpoint
	Vector3f rate_setpoint = eq.emult(_proportional_gain);

	// Feed forward the yaw setpoint rate.
	// yawspeed_setpoint is the feed forward commanded rotation around the world z-axis,
	// but we need to apply it in the body frame (because _rates_sp is expressed in the body frame).
	// Therefore we infer the world z-axis (expressed in the body frame) by taking the last column of R.transposed (== q.inversed)
	// and multiply it by the yaw setpoint rate (yawspeed_setpoint).
	// This yields a vector representing the commanded rotatation around the world z-axis expressed in the body frame
	// such that it can be added to the rates setpoint.
	if (std::isfinite(_yawspeed_setpoint)) {
		rate_setpoint += q.inversed().dcm_z() * _yawspeed_setpoint;
	}

	// limit rates
	for (int i = 0; i < 3; i++) {
		rate_setpoint(i) = math::constrain(rate_setpoint(i), -_rate_limit(i), _rate_limit(i));
	}

	return rate_setpoint;
}

2.APM transplant PX4 algorithm：

void AC_AttitudeControl::nonlinear_control_quat(Quaternion& att_to_quat, const Quaternion& att_from_quat, Vector3f& att_diff_angle, float& thrust_vec_dot)
{

	//旋转从机体坐标系到惯性坐标系---- rotation from the target body frame to the inertial frame.
    Matrix3f att_to_rot_matrix;
    //得到惯性坐标系下的方向余弦矩阵，注意att_to_quat已经进行归一化处理了
    att_to_quat.rotation_matrix(att_to_rot_matrix); //目标到实际的地理坐标系下，目标四元数
    //获取垂直矢量
    Vector3f att_to_thrust_vec = att_to_rot_matrix * Vector3f(0.0f, 0.0f, 1.0f); //3*3和3行1列得到地理坐标系下的z轴数据三行一列
    //旋转当前机体坐标系到惯性坐标系-----rotation from the current body frame to the inertial frame.
    Matrix3f att_from_rot_matrix;
    //得到实际的欧拉角的四元数旋转矩阵
    att_from_quat.rotation_matrix(att_from_rot_matrix);  //实际四元数
    //得到实际的垂直信息,n【az,bz,cz】
    Vector3f att_from_thrust_vec = att_from_rot_matrix * Vector3f(0.0f, 0.0f, 1.0f);//得到三行一列

    Quaternion qd_red(att_from_thrust_vec, att_to_thrust_vec);

    thrust_vec_dot = 2.0f * acosf(qd_red.q1);
    if(fabsf(qd_red.q2) > (1.f - 1e-5f) || fabsf(qd_red.q3) > (1.f - 1e-5f))
    {
    	qd_red = att_to_quat;
    }
    else
    {
    	/* transform rotation from current to desired thrust vector into a world frame reduced desired attitude */
    	qd_red *= att_from_quat;  

    }

	/* mix full and reduced desired attitude */
    Quaternion q_mix = qd_red.inverse() * att_to_quat;

    if(q_mix.q1 < 0)
    {
	 q_mix = q_mix * -1.0f;
    }
	/* catch numerical problems with the domain of acosf and asinf */
	q_mix.q1 = constrain_float(q_mix.q1, -1.f, 1.f);
	q_mix.q4 = constrain_float(q_mix.q4, -1.f, 1.f);
	// _yaw_w 【0,1】
	Quaternion qd = qd_red * Quaternion(cosf(_yaw_w  * acosf(q_mix.q1)), 0, 0, sinf(_yaw_w  * asinf(q_mix.q4)));

	/* quaternion attitude control law, qe is rotation from q to qd */
	Quaternion qe = att_from_quat.inverse() * qd;

	att_diff_angle.x = 2.0f *  qe.q2;
	att_diff_angle.y = 2.0f *  qe.q3;
	att_diff_angle.z = 2.0f *  qe.q4;

	if(qe.q1 < 0)
	{
		att_diff_angle = att_diff_angle * -1.0f;
	}
}

lthall, 2024-02-22:
In what way do believe it is superior?

lixiaowei123, 2024-02-22:
> In what way do believe it is superior?

[image]

A friend has verified that your implementation method is indeed theoretically supported,

lthall, 2024-02-22:
What do mean by "theoretically supported"?

lixiaowei123, 2024-02-22:
> What do mean by "theoretically supported"?

You can verify it by flying, but I believe that the attitude error calculation of Ardupilot has no formula or theoretical basis and cannot be written arbitrarily based on your own experience

lthall, 2024-02-22:
I don't understand what you mean by that. We could not write the calculation as code unless we could represent it as a formula with a solid theoretical basis.

Our algorithm is a really simple two rotation solution.

I must be misunderstanding your statement.  Can you try to rephrase your answer.

lixiaowei123, 2024-02-22:
[https://flyingmachinearena.org/wp-content/publications/2018/breTCST18.pdf](url)
Thank you very much. You can take a look at this paper

lthall, 2024-02-22:
What is in that paper that I should look at that answers my question?

lthall, 2024-02-22:
I would suggest that you post questions and observations on the discussion forums. You will receive a much more positive response there as it is the appropriate forum.

I get the impression you are a young person that is working hard to develop an understanding of multirotor dynamics. I can see there are a lot of factors and issues that you don't understand or appreciate. However your misuse of the git issues as a question forum does not inspire people to assist you in your educational journey.

closed 2024-02-22 by lthall

lixiaowei123, 2024-02-22:
This paper has a rigorous theoretical basis for calculating attitude errors, which seems more reasonable and has been validated in PX4. Based on your previous response in this regard, you have designed it based on your own experience and have not referred to the paper or anything else. I hope our code implementation has better theoretical basis。
[image]

lthall, 2024-02-22:
Publishing the equations does not provide a theoretical basis. You also seem to be under the impression that to do good work you must copy somebody else. 

If I had taken the time to write up and publish the algorithm I design it would make no difference to the performance or theoretical basis.

I would also point out that good engineering solutions are very often found independently by many different engineers without looking to others. This is because they are simply the logical conclusion when you understand the issus.