ArduPilot/ardupilot #26295: Copter-4.5.0-beta2: Neopixel LEDs out-of-sync

rmackay9, 2024-02-23:
If multiple neopixel LEDs are connected they may show different colours from each other in 4.5.0 while in 4.4.4 they all show the same colour.

Below is a video and I hope to get more details of the setup from the user.
https://github.com/ArduPilot/ardupilot/assets/1498098/f0a9a517-1dc4-4099-85e6-80ee0af6d7d2

andyp1per, 2024-02-23:
@rmackay9 if you are using CubeOrange then you really need to use a pull-up on the output. I have had this discussion on and off with @bugobliterator. Not saying this is the issue, but CubeOrange is definitely special electrically.

bugobliterator, 2024-02-23:
https://discuss.cubepilot.org/t/rgb-neopixel-led-wiring-and-configuration/12129/8?u=sidbh Reference to the discussion where I answered this query.

[image]

The reason that NeoPixel LEDs don’t function on Cube is because they require > 0.7*Vcc signal line, Aux pins put out 3.3V and generally power tends to be solid 5V.
0.7*5 = 3.5V Reference 

If Neopixel are working without any modification, it is actually more of a cause of concern, as that indicates your 5V are not actually 5V but something like 4.7V, and possible symptom of this is brown-out on other hardware that require solid 5V for instance ESC or Autopilot power system etc...

tridge, 2024-03-08:
I think it is more likely a slight change in clocking. I got the same symptoms (very similar flashing) on a G4 periph, and fixed it by overriding STM32_PPRE1 so we got a different peripheral clock on the timer

andyp1per, 2024-04-27:
Fixed by https://github.com/ArduPilot/ardupilot/pull/26899

closed 2024-04-30 by tridge