ArduPilot/ardupilot #26325: sitl simulink ans json simulation, matlab is always busy and does see anything in the command window

aogrcs, 2024-02-26:
I test the master branch, matlab r2022b, start the command line
./Tools/autotest/sim_vehicle.py -v ArduCopter -M json:127.0.0.1 --comsole --map
just follow the vedio in this link
https://ardupilot.org/dev/docs/MATLAB-Simulation.html
but the command line doese see anythin except busy
could anyone give me some suggestion, thanks
sincerely

aogrcs, 2024-02-27:
> Does normal SITL work correctly? Did you try `./Tools/autotest/sim_vehicle.py -v ArduCopter -f json:127.0.0.1 --comsole --map`? What do you see in the SITL window? What do you see in MATLAB? Are MATLAB and SITL on the same machine?

Hi, this is the sitl window
[image]
I use wsl2 and ubuntu 22.04, thanks

IamPete1, 2024-02-27:
Looks like the ArduPilot side is OK. Which version of MATLAB? What output do you get there? Is it on the same machine?

aogrcs, 2024-02-27:
> Looks like the ArduPilot side is OK. Which version of MATLAB? What output do you get there? Is it on the same machine?

Hi, I use Matlab 2022b, there is no output, and they are on the same machine, I use ifconfig, this is the output
[image]

IamPete1, 2024-02-27:
I have just tested on master and MATLAB 2022b and it works for me on with `127.0.0.1` WSL1 but not WSL2, it I put in my local IP `192.168.1.x` then it works on WSL2 as well. I suspect its just a networking thing in WSL2.

[image]

aogrcs, 2024-02-28:
> I have just tested on master and MATLAB 2022b and it works for me on with `127.0.0.1` WSL1 but not WSL2, if I put in my local IP `192.168.1.x` then it works on WSL2 as well. I suspect its just a networking thing in WSL2.
> 
> ![image](https://private-user-images.githubusercontent.com/33176108/308314249-2ed97df4-a4ce-4ac9-80a7-102d742ded98.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwODE3NTgsIm5iZiI6MTcwOTA4MTQ1OCwicGF0aCI6Ii8zMzE3NjEwOC8zMDgzMTQyNDktMmVkOTdkZjQtYTRjZS00YWM5LTgwYTctMTAyZDc0MmRlZDk4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI4VDAwNTA1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIyMTUyMzFlNjdlYzgzMzlhOTZhYjgwZTI1ODU1ZTI5MWYwMjdkNjE0OTg5OTVhZjk2MjI2YThiZmQzMzZmYjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.T2XRF2ZbEPHwF9xMvi1FpmuUVQw4RUY3LdnitWtuHfg)

Thanks, I use this command
ip route | grep default | awk '{print $3}'
getting my host ip using for json and it works

sincerley

closed 2024-02-28 by aogrcs