ArduPilot/ardupilot #26340: Plane: TECS does not use PTCH_TRIM_DEG TRIM_PITCH_CD (PTCH_TRIM_DEG) for flights without pitot or TECS synthetic airspeed

fredowski, 2024-02-27:
## Bug resport

**Issue details**
I have a strange behaviour flying plane 4.4.0-beta3 without pitot and without synthetic airspeed in TECS. I have set TRIM_PITCH_CD = -800, TRIM_THROTTLE = 80, TRIM_ARSPD_CM = 2000. However, during level flight the plane settles to an average throttle of 67, a pitch of -4 degrees and an average speed of 17 m/s. 

[image]

I expect an average throttle of 80 percent during level flight, but the resulting throttle is constantly below that. 

**Reason**
It could be explained by the fact that the pitch in update_throttle_without_airspeed: https://github.com/ArduPilot/ardupilot/blob/b271a20e9799db97c05bdffcd16ccbd16172da66/libraries/AP_TECS/AP_TECS.cpp#L848 does not use pitch with respect to pitch_trim_cd. However in the following line https://github.com/ArduPilot/ardupilot/blob/b271a20e9799db97c05bdffcd16ccbd16172da66/libraries/AP_TECS/AP_TECS.cpp#L849 the pitch_trim_cd is subtracted. If I assume that _ahrs.pitch is simply taken as is in the following throttle computation, then in line 858 the throttle would be computed to 80 + (0 - 80) * -4/-25 = 67 which is the actual achieved average throttle. However, I do not understand that behaviour because in line 849, the correction for pitch_trim_cd is made. 

It turns out that TRIM_PITCH_CD is not considered and aparm.pitch_trim_cd is always 0. That problem was found by by @timtuxworth in https://github.com/ArduPilot/ardupilot/pull/25768#issuecomment-1855041344. It seems that the variable aparm.pitch_trim_cd is never set. The variable was introduced in #22191. 

I described the problem in the first place here: https://github.com/ArduPilot/ardupilot/pull/25071#issuecomment-1967382494 where @IamPete1 pointed me to the root cause.

@timtuxworth already proposed a fix https://github.com/ArduPilot/ardupilot/pull/25768#discussion_r1427177386 during his rework of the centi degrees parameter changes and @IamPete1 mentioned that that should be done outside the parameter rework as it changes the flight behaviour.

Maybe the changes should be done according to the ideas https://github.com/ArduPilot/ardupilot/pull/25071/commits/0f698601b9ecb310b458ad779539dd438b386701 from @priseborough  as part of #25071, i.e. that the corrected pitch value is also used for initialization.

**Version: 4.4.0-beta3**
**Airframe: Fixed Wing**
**Flight Controller: Matek**
**Log File**
I attach the .bin logfile from the flight.

[239-1-20230817-Jacky.BIN.zip](https://github.com/ArduPilot/ardupilot/files/14423613/239-1-20230817-Jacky.BIN.zip)

IamPete1, 2024-02-27:
Can you test on 4.5? There have been a few TECS changes since 4.4.

fredowski, 2024-02-27:
Currently it is pretty cold, so I cannot test now. I have seen that the parameter renaming is also in 4.5, but the ptch_trim_deg / trim_pitch has been removed in TECS https://github.com/ArduPilot/ardupilot/blob/ab68066e9c7a671b6a032289d5eda382b3d3245b/libraries/AP_TECS/AP_TECS.cpp#L862

in order to not change the flight behaviour. That was your idea to keep the flight behaviour identical for the moment, no?

IamPete1, 2024-02-27:
> That was your idea to keep the flight behaviour identical for the moment, no?

That particular change will keep the flight behavior. But there are several other TECS changes between 4.4 and 4.5.

If you could come up with a "recipe" to replicate the issue in the simulator that would be very useful.

fredowski, 2024-02-29:
I did a SITL simulation on arduplane master as of 29.2.2024 (3926e06508ffa). I changed ARSPD_USE to 0. During flight I changed PTCH_TRIM_DEG from 0 to 10 to -10 and back to 0. I see the same behaviour in that sense that the resulting throttle during level flight does not depend on the setting of the PTCH_TRIM_DEG. The TECS.ph and ATT.DesPitch values are offset by the PTCH_TRIM_DEG value with respect to the ATT.pitch.

[image]

I plotted the logfile of the SITL simulation. You can see that the throttle always settles at the same 42% value regardless of the  PTCH_TRIM_DEG value. The ATT.pitch settles at -3.5 degrees. When PTCH_TRIM_DEG is changed during flight, there is a temporary altitude error. The TECS integrator will settle at another value, the TECS.ph and ATT.DesPitch are then offset by PTCH_TRIM_DEG with respect to ATT.pitch. TRIM_THROTTLE is 50%.

My understanding of what PTCH_TRIM_DEG should do is that setting the parameter should be equivalent to a different calibration on the ground with a different attitude. This translates to two effects:

a) If I lift the nose about 5 degrees during calibration, then this should be the same as if I set PTCH_TRIM_DEG to 5 degrees when the calibration was done flat. The horizon should show 0 pitch when the plane is actually 5 degrees up.

b) The setpoint for the levelflight in TECS is changed by PTCH_TRIM_DEG

The current behaviour does not achieve a) nor b).

Maybe it would be an idea to change the behaviour of TECS such that TRIM_THROTTLE is achieved during level flight and the resulting pitch just depends on the plane. Then PTCH_TRIM_DEG would not be required. Now the TECS controller settles on a flyable combination of pitch and throttle. I want to change the speed of the plane by simply adjusting TRIM_THROTTLE.

Hwurzburg, 2024-02-29:
if changing pitch trim has no effect on cruise throttle and speed steady state then there is a bug...see https://studio.youtube.com/video/B6nqJZMqsuQ/edit for how TECS is supposed to work...and has historically

fredowski, 2024-02-29:
Sorry, I cannot see your video. That requires an account login.

Hwurzburg, 2024-02-29:
made it public...shoud have been able to see with that link...I have been simming with master on plane and changing ptch_trim or trim_throttle affects cruise speed set point/throlle pitch as expected....the sim plane has a very flat power curve so large changes in ptch_trim_deg or throttle are neede to move the cruise speed significantly

fredowski, 2024-02-29:
Sorry - but it still asks for a login. Can you send the public link?

Hwurzburg, 2024-02-29:
https://youtu.be/k2c6o-UCmps

Hwurzburg, 2024-02-29:
https://youtu.be/B6nqJZMqsuQ

fredowski, 2024-02-29:
Thanks - video works now.

fredowski, 2024-02-29:
Thanks for the video. You are aware that usage of the PTCH_TRIM_DEG (the new TRIM_PITCH_CD) parameter has been removed from the current TECS code because the previous parameter was never set? See: https://github.com/ArduPilot/ardupilot/commit/b1c85113862e97a3849399b2c63d192c09c0ee1c

Changing TRIM_THROTTLE has an effect on actual throttle, but throttle never reaches TRIM_THROTTLE, except the plane flies with a pitch of 0 degress with the airspeed that is the result of TRIM_THROTTLE.

timtuxworth, 2024-02-29:
I'm no expert in this area, I just found the bug where the value used by TECS was never being set. In the discussion about how to fix the bug, there was pretty much a consensus that TECS should not know about the pitch_trim value and that the reference to it was not a good idea.
As it works now, pitch_trim definitely affects the flying attitude of the plane, I've tested it and I know it works. 
Beyond that - you might have to explain the problem you have in a different way, the problem you are seeing doesn't seem to be clear.

fredowski, 2024-02-29:
The problem that I see during flight in auto mode is that PTCH_TRIM_DEG does not affect the flight. It is the same problem that is visible in the simulator. As explained in the video PTCH_TRIM_DEG and TRIM_THROTTLE must match but today you have effective PTCH_TRIM_DEG=0. I want to change the speed by modifying TRIM_THROTTLE but changing the value does not result in a level flight with that throttle value.

fredowski, 2024-03-02:
I did another SITL simulation at commit https://github.com/ArduPilot/ardupilot/commit/d335e717492aa1b979652f8e9f39405d03224682 which is the last commit before https://github.com/ArduPilot/ardupilot/commit/0840bf5a2190a7b397367539dd3941994fe5aed1 where the TECS throttle computation is changed. At that commit TRIM_PITCH_CD influences the flown throttle and pitch. 

[image]

The graph shows a SITL simulation at level flight with ARSPD_USE=0, i.e. without airspeed sensor. The first part is with TRIM_PITCH_CD=-1000 and the second part with TRIM_PITCH_CD=+1000. You can see that flown throttle, airspeed and pitch are affected by the parameter setting.

I guess the reason why TRIM_PITCH_CD does not have an effect anymore after commit https://github.com/ArduPilot/ardupilot/commit/0840bf5a2190a7b397367539dd3941994fe5aed1 is that the trim_pitch comes via _ahrs indirectly and then has an effect on _pitch_dem. But with the new commit only a highpassed version of _pitch_dem is used for throttle, i.e. the TRIM_PITCH_CD will not have an effect on throttle because it is filtered away. Commit https://github.com/ArduPilot/ardupilot/commit/0840bf5a2190a7b397367539dd3941994fe5aed1 tries to use the pitch_trim, but because that value is never set in the aparm parameter block it is permanently zero.

@priseborough: I think the PTCH_TRIM_DEG is necessary for correct TECS behaviour. Otherwise I cannot set a correct cruise operating point where PTCH_TRIM_DEG and TRIM_THROTTLE are actually flown. Regarding implementation: I think AP_TECS has no access to the global parameters. What would be the best approach to bring the parameter in?

I had quite some problems setting cruise speed because I flew PX4 1.8.2 some time ago where the throttle was derived by height demand rate such that at level flight always TRIM_THROTTLE was reached. I think this equivalent to the TECS very first commit 

https://github.com/ArduPilot/ardupilot/blob/5b0129e02b4c816becae32d572ad5bf0030ff893/libraries/AP_TECS/AP_TECS.cpp#L543

I made a hack https://github.com/ArduPilot/ardupilot/compare/master...fredowski:ardupilot:tecs-fix-ptch-trim to bring back the old behaviour and derived the throttle from _hgt_rate_dem and did a SITL simulation with that. 
[image]
The graph shows the SITL simulation where throttle is only controlled by _hgt_rate_dem. I flies nicely and TRIM_THROTTLE is reached during level flight. At the 50% throttle the flown pitch angle is -3.2 degrees.

With that hack upwinds or downwinds will not change the throttle. 
[image] 
The graph shows a SITL simulation with downwind of 5 m/s. During level flight the pitch is increased to 12.3 degrees still flying at 50% throttle. During descent the plane stalls at an airspeed of 7 m/s. 

This behaviour made it straightforward to set a cruise throttle and change the speed during flight because I only had to set TRIM_THROTTLE and the plane flew with that throttle during level flight. However not adapting the throttle during strong gusts is also not nice. I wonder if it is possible to estimate the PTCH_TRIM_DEG which should only be a function of the plane and to differ it from the influence of up- or downwinds. 

I like the idea but it goes beyond the current problem to bring PTCH_TRIM_DEG back to TECS.

Hwurzburg, 2024-03-02:
you are correct there is an issue introduced in late 4.4....working now with Tridge...he is doing a patch as we speak ...I will flight test tomorrow weather permitting and it willl backport to 4.4 when tested

fredowski, 2024-03-02:
Thanks for the fix and the plan to test it! I just created https://github.com/ArduPilot/ardupilot/pull/26377 . My patch moved pitch_trim to the aparm block but otherwise it is identical. I tried that in SITL:
[image]
The graph shows the SITL simulation with the fixed TECS. The throttle changes when PTCH_TRIM_DEG is modified.

closed 2024-03-03 by tridge

Hwurzburg, 2024-03-03:
test flown...I could take a FBWA level point, set the PTCH_TRIM_DEG and TRIM_THROTTLE for that point and get that speed in CRUISE....PTCH_TRIM_DEG being in a TECS calculation was confusing to me until I realized that Paul was trying to obtain ATT.DesPitch to calculate Throttle diff from TRIM_THROTTLE and he fetched the AHRS pitch which is demanded pitch + pitch trim, so it needs to be removed from ahrs pitch to get demanded pitch to drive the throttle adjustment

fredowski, 2024-03-03:
Good to hear! Thank you for testing!