ArduPilot/ardupilot #26357: Copter automatically turns north when switching to GUIDED_NOGPS

ShevaDavid, 2024-02-29:
## Bug report

**Issue details**
This issue follows up the discussion with Leonard
https://discuss.ardupilot.org/t/set-attitude-target-with-guided-no-gps-lack-of-documentation/111845/24
When I switch to GUIDED_NOGPS, the copter automatically turns north, because this is how attitude quaternion is initialized (1, 0, 0, 0) in angle_control_start method. So I just replaced it with zero quaternion (0, 0, 0, 0) and now there is no automatic north rotaion.

**Version**
4.5

**Platform**
[  ] Copter

rmackay9, 2024-02-29:
Hi, thanks for the report. I've added this to the Copter-4.5 issues list although I guess the issue also exists in 4.4.  Txs again

lthall, 2024-03-06:
PR #26407

lthall, 2024-03-31:
This has now been fixed.

closed 2024-03-31 by lthall