ArduPilot/ardupilot #26358: Sending unknown message (37) error with custom server build of 4.5.0-beta2

DavideLentini, 2024-02-29:
## Bug report

**Issue details**

Hi all , I programmed the drone with firmware 4.5.0-beta2 using these definitions , to remove the items I don't need/interest . 
Loading the firmware , the drone seems to work without problems , but it returns every 2/3 seconds this message : 

[image]

Loading clean firwmare 4.5.0-beta2 , this problem does not occur . It could be caused by my definitions but without a definition of the error I wouldn't really know where to look . Thanks

**Version**
4.5.0-beta2

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2024-02-29:
It was rangefinder, we forgot to tell it not to try sending the message if rangefinder is not there. fix here: https://github.com/ArduPilot/ardupilot/pull/26361

IamPete1, 2024-03-05:
Fixed by https://github.com/ArduPilot/ardupilot/pull/26361

closed 2024-03-05 by IamPete1