ArduPilot/ardupilot #26370: Means to trigger RTL or Land based on battery data

Neel5897, 2024-03-01:
## Feature request ## 

**Is your feature request related to a problem? Please describe.**
We would like BMS to send commands for RTL or E-Land based on battery data. For example, in an event of overcurrent or overtemperature way above the acceptable threshold, the UAV should perform an emergency landing sequence 

**Describe the solution you'd like**
One idea would be to use the status flags in BatteryInfo.h or BatteryContinuous.h If any of these flags are triggered it can do the appropriate tasks to initialise an RTL or Land

**Describe alternatives you've considered**
As of now, I'm initializing RTL or Land by changing the parameters BATT_FS_LOW_ACT and BATT_FS_CRIT_ACT in mission planner. But these can only perform the RTL or land based on voltage and SOC. 

**Platform**
[  ] Battery management system (BMS)
[  ] Copter
[  ] Plane
[  ] DSDL

rmackay9, 2024-03-01:
Hi @Neel5897,

For more complicated battery failsafes, a Lua script might be a good idea.  So the battery provides the data but the Lua script makes the decisions on what actions to take.

By the way, where are the BatteryInfo.h or BatteryContinuous.h that you've mentioned?

Neel5897, 2024-03-05:
Hi @rmackay, 
I've tried my luck at lua scripting but its been unsuccessful so far. If there's any example codes or references, that could be of great help. 

The batteryInfo.h is in the path uavcan>equipment>power of droneCAN DSDL. And the batteryContinuous.h is in ardupilot>equipment>power.

peterbarker, 2025-05-11:
@Neel5897 scripting is probably a good way forward for advanced failsafe handling.

For actually getting extra information from the battery - I find it *very* strange that ArduPilot isn't using the the flags from `uavcan_equipment_power_BatteryInfo`.  We'd want to inform our health state based on that information if you want to supply it, I think.

The BatteryContinuous stuff is still a WIP AFAIK.  We have definitions but no implementation anywhere, AFAIK.

peterbarker, 2025-05-11:
@Neel5897 is this still something you're chasing?