ArduPilot/ardupilot #26386: Add Raspberry Pi 5 support

patrickelectric, 2024-03-04:
## Feature request

**Is your feature request related to a problem? Please describe.**
There is no Raspberry Pi 5 support on the project.

**Describe the solution you'd like**
Be able to use the HAL that currently exist for Raspberry Pi 0/1/2/3/4 with the Pi 5.

**Describe alternatives you've considered**
There is none 🤷

**Platform**
[x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
It appears that the Raspberry Pi 5 is based over the RPI1 and it's not compatible with the current software architecture used by the others PIs. In the end, is not just a matter of changing the base peripherals address.

ES-Alexander, 2024-03-11:
For tracking purposes, there's [a forum thread](https://discuss.bluerobotics.com/t/pi-5-with-navigator/15903) of ArduSub users requesting/discussing this.

BoKoppel, 2024-05-09:
I dont know if this fix helps evolving BlueOS on Pi5.
I use the fix for plotting using openplotter, the fix switches the Pi os to use X11 instead of Wayland:
https://raspberrypi.stackexchange.com/questions/144866/can-i-use-x11-on-the-new-64-bit-os-instead-of-wayland

PR #27026 (2024-05-13): AP_HAL_Linux: Add Raspberry Pi 5 GPIO support
Fix #26386

Adds support to RP1 coprocessor.
closed 2024-05-13 by Williangalvani