ArduPilot/ardupilot #26404: Unable to command velocity in GUIDED MODE

CrucadingCodes, 2024-03-05:
## Bug report

Unable to command velocity commands (in GUIDED mode)  through  SET_POSITION_TARGET_GLOBAL_INT/ SET_POSITION_TARGET_LOCAL_NED

I'm trying to command the bird's velocity in SITL through mavutil using the above messages. However these commands/messages don't seem to be sending, In comparision, I am able to send the SET_ATTITUDE_TARGET message and send attitude commands(in GUIDED Mode). Is there any way I can command velocity (for Copter) in GUIDED Mode, am I missing something? 

**Version**
 V4.6.0-dev 

**Platform**

[  ] Copter

**Airframe type**
quad

**Hardware type**
using SITL 

[**Logs**]
[velocity_control.log](https://github.com/ArduPilot/ardupilot/files/14494073/velocity_control.log)

adding my code as well 

`

    import time
    import math

    from pymavlink import mavutil
    import numpy as np

    def set_target_velocity(velocity_x, velocity_z):
   

   
    
        msg = master.mav.set_position_target_global_int_encode(
                0,       # time_boot_ms (not used)
                0, 0,    # target system, target component
                mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT_INT, # frame
                mavutil.mavlink.POSITION_TARGET_TYPEMASK_X_IGNORE |
                mavutil.mavlink.POSITION_TARGET_TYPEMASK_Y_IGNORE |
                mavutil.mavlink.POSITION_TARGET_TYPEMASK_Z_IGNORE |
                mavutil.mavlink.POSITION_TARGET_TYPEMASK_VY_IGNORE |
                mavutil.mavlink.POSITION_TARGET_TYPEMASK_AX_IGNORE |
                mavutil.mavlink.POSITION_TARGET_TYPEMASK_AY_IGNORE |
                mavutil.mavlink.POSITION_TARGET_TYPEMASK_AZ_IGNORE |
                mavutil.mavlink.POSITION_TARGET_TYPEMASK_FORCE_SET |
                mavutil.mavlink.POSITION_TARGET_TYPEMASK_YAW_IGNORE |
                mavutil.mavlink.POSITION_TARGET_TYPEMASK_YAW_RATE_IGNORE,
                0, # lat_int - X Position in WGS84 frame in 1e7 * meters
                0, # lon_int - Y Position in WGS84 frame in 1e7 * meters
                0, # alt - Altitude in meters in AMSL altitude(not WGS84 if absolute or relative)
                # altitude above terrain if GLOBAL_TERRAIN_ALT_INT
                velocity_x, # X velocity in NED frame in m/s
                0, # Y velocity in NED frame in m/s
                velocity_z, # Z velocity in NED frame in m/s
                0, 0, 0, # afx, afy, afz acceleration (not supported yet, ignored in GCS_Mavlink)
                0, 0)    # yaw, yaw_rate (not supported yet, ignored in GCS_Mavlink) 
    
    
    
        for x in np.arange(0.0, 100, 0.1):
        
      
           master.mav.send(msg)

           time.sleep(0.1)
        
       
        
        
    master = mavutil.mavlink_connection('127.0.0.1:14550')

    master.wait_heartbeat()

    master.set_mode('GUIDED')
    print("Mode changed --")
    time.sleep(1)

    print("commanding")

    set_target_velocity(velocity_x = 10, velocity_z = 2) `

lthall, 2024-03-05:
This is not the right place to ask for help. These messages are used in many of our automated checks.

This is what I use when testing this in SITL

module load message
message SET_ATTITUDE_TARGET 0 1 0 7 [1,0,0,0] 0 0 0 0.5
message SET_ATTITUDE_TARGET 0 1 0 32 [1,0,0,0] 0 0 0 0.5
message SET_ATTITUDE_TARGET 0 1 0 39 [1,0,0,0] 0 0 0 0.5
message SET_ATTITUDE_TARGET 0 1 0 39 [0.9659,-0.2588,0,0] 0 0 0 0.5
message SET_ATTITUDE_TARGET 0 1 0 7 [0.9659,-0.2588,0,0] 0 0 0 0.5
message SET_ATTITUDE_TARGET 0 1 0 128 [1,0,0,0] 1 0 0 0.5
message SET_POSITION_TARGET_GLOBAL_INT 0 1 0 11 3576 -353641760 1491597880 20 0 0 0 0 0 0 0 0
message SET_POSITION_TARGET_GLOBAL_INT 0 1 0 11 3576 -353632578 1491652316 20 0 0 0 0 0 0 0 0
message SET_POSITION_TARGET_GLOBAL_INT 0 1 0 11 3576 363215670 1386596701 20 0 0 0 0 0 0 0 0
message SET_POSITION_TARGET_LOCAL_NED  0 1 0 1 448 0 0 0 0 0 -5 0 0 0 0 0
message SET_POSITION_TARGET_LOCAL_NED  0 1 0 1 3135 0 0 0 0 0 0 1 2 3 0 0
message SET_POSITION_TARGET_LOCAL_NED  0 1 0 1 3135 0 0 0 0 0 0 1 2 3 0 0

You should be able to alter those examples to get what you need.

closed 2024-03-05 by lthall

rmackay9, 2024-03-06:
Hi @CrucadingCodes,

In case it helps, here's a developer wiki page which describes how to use the SET_POSITION_TARGET_LOCAL_NED along with examples that can be copy-pasted into MAVProxy.

https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html#set-position-target-local-ned

CrucadingCodes, 2024-03-06:
> Hi @CrucadingCodes,
> 
> In case it helps, here's a developer wiki page which describes how to use the SET_POSITION_TARGET_LOCAL_NED along with examples that can be copy-pasted into MAVProxy.
> 
> https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html#set-position-target-local-ned

Hi @rmackay9  I tried exactly that, and it works on MavProxy, but not through my script (I am trying to command it through a custom script), even when I have followed the steps as well within the script. Is it possible that I am unable to access the "message" module or it's not loadingf ? if yes how can I access/load  it through pymavlink's mavutil

rmackay9, 2024-03-06:
@CrucadingCodes,

Trying to run it on SITL sounds like a good idea.  Surely there is some small difference in the message format between what the script sends and what MAVProxy's message module sends.  You may also be able to see the difference by pulling up the mavlink inspector.  Immediately after sending the message it should appear there (in the mavlink inspector).  I forget how to use the mavlink inspector with MAVProxy but MP and QGC have them.

CrucadingCodes, 2024-03-06:
> @CrucadingCodes,
>
> Trying to run it on SITL sounds like a good idea. Surely there is some small difference in the message format between what the script sends and what MAVProxy's message module [...]

@rmackay9
The thing that baffles me is that I've tried sending both SET_ATTITUDE_TARGET and SET_POSITION_TARGET_LOCAL_NED/SET_POSITION_TARGET_GLOBAL_INT, through a custom script and ran it on SITL, the SET_ATTITUDE_TARGET message definitely sends and I can see the commanded values on the simulation, it's not the same with the other and the only logical explanation I can come to at this point is that the message is unable to command (confirmed this through log values, will try to double confirm through Mavlink Inspector). I'm using a Linux based System and somehow I always end up loosing the connection to QGC(don't know why), when i run the SITL(through the script). Usually on MavProxy, we  just get  COMMAND_ACK indication. (can't really confirm anything through that).

CrucadingCodes, 2024-03-06:
@rmackay9 
I found the issue, it was on how the message was being send, somehow through the IGNORE bit isn't being able to decode. So I sent the message how it's mentioned in the documents, using the type mask (0b110111000111 -> for enabling speeds only) instead of  IGNORE , works perfectly well and would recommend using that only. 
Thanks for your support though.

PR #26407 (2024-03-25): Copter: Guided Angle: Initialize yaw to current yaw.
This fixes #26404.

I am simply initialising the attitude to the current heading so it does not rotate when it doesn't receive an initial attitude like normal guided attitude mode.