ArduPilot/ardupilot #26416: Brake mode parameters for harder acceleration

GalacticGreg, 2024-03-06:
## Feature request

**Is your feature request related to a problem? Please describe.
I teach students how to 3d print drone frames and we fly them outside. I have brake mode on a switch but it doesn't brake aggressively enough.

**Describe the solution you'd like**
I want students to be able to brake quickly if they run into a problem. The max acceleration for brake mode is less than what I have loiter mode set to, so braking feels sluggish.

**Describe alternatives you've considered**
Begging a developer to implement the changes. Thanks for all your hard work btw!

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

IamPete1, 2024-03-06:
Looks like its hard coded to 7.5m/s^2 here:

https://github.com/ArduPilot/ardupilot/blob/690596580a0650e4f7f23069999ac8899a3d4096/ArduCopter/mode_brake.cpp#L13-L14

What acceleration are you using in loiter/auto?

It seems a reasonable change to use the max of the existing default and the position controller value. I'm a bit surprised were not just using that value all the time.

GalacticGreg, 2024-03-06:
Clearly my perception is off, my Loit_acc_max is 5m/s^2 with a jerk of 5m/s^3.

Despite that I'd like to be able to configure it to come to a screeching halt, haha.

It's own parameter of brake_angle_max and brake_accel_max/brake_jerk would be best though. I'd like it to stop harder than I allow the students to drive.

It must use Angle_max as the angle limit?

IamPete1, 2024-03-06:
> It must use Angle_max as the angle limit?

or `PSC_ANGLE_MAX` if its set. If you have a log we should be able to work out what the limiting factor is.

GalacticGreg, 2024-03-08:
Next week is spring break for me so I'll do some texting then and report back.

GalacticGreg, 2024-03-13:
[Brake from Loiter.zip](https://github.com/ArduPilot/ardupilot/files/14591709/Brake.from.Loiter.zip)
Here is a log of the drone stopping from loiter mode after a full roll, followed by a couple of stops using brake mode from full roll.

My max. angles are set to 40 degrees. You can see that loiter stops much quicker than brake mode.

This log is sufficient I believe to demonstrate the issue with brake mode, but you can see that it is a partial log. I just spent a few hours trying to get a good full log but I'm having problems with "bad logging" errors.

Perhaps this is a separate issue with 4.5beta2? I've tried 2 different 32GB Sandisk class 10 SD Cards freshly formatted to FAT32 and I get bad logging errors shortly after taking off. When I notice the bad logging errors I land and when I try to rearm, my ESC's reboot and I get the following errors. Once was the 0x4000 and a separate time was 0x8000.
Internal error 0x4000
internal error 0x8000 1:369 main loop
AP-Logger: stuck thread (disk_space_avail)

Let's focus on brake made for now though. Any ideas why loiter brakes faster?

IamPete1, 2024-03-13:
If we just plot GPS speed there is not a huge amount in it. 

[image]

They seem to slow down about the same to start with to (4 m/s or so), but loiter comes to a harder stop where as brake slows more gradually after that. 

Plotting the desired accelerations gives lovely straight lines when break is first entered. So the initial slow down is jerk limited.

[image]
 
You could increase the jerk limit to speed up that initial slow down. `PSC_JERK_XY`.

Once it hits the curved section that is a factor of the rest of the input shaping trying to bring it to a gradual stop. Higher accel limits and gains will make that faster. 

I have not done the position controller yet, but I have made a little tool to visualize what the input shaping does in the angle controller. The concepts are much the same.

https://firmware.ardupilot.org/Tools/WebTools/KinematicTool/

lthall, 2024-03-13:
I would suggest trying:
PSC_JERK_XY	15 to 30

GalacticGreg, 2024-03-14:
Thanks, I'll change that. Just to be clear, that will affect the first part of the deceleration right? That will help for sure.

It is that last part of the slow down that is the bigger problem. It doesn't look too different looking at the graph, but once it gets to that last 4 m/s it drifts about 15-20 feet and feels more like "coast to a stop" mode instead of brake mode, haha. Ideally it would come to a stop quicker than loiter braking. If one of my kids is in alt hold, heading for the tree line, I'd like that panic brake switch to stop on a dime.

GalacticGreg, 2024-03-14:
I just took a look at that tool you linked. If I turned off input shaping (ATC_RATE_FF_ENABLE) would I get the desired braking effect do you think?

Just a thought, I'm not sure I'd like the result during the rest of the flight...

lthall, 2024-03-14:
It will increase the slope on both the entry and exit from the break mode. I have tried to highlight this in the image below with the red and green lines.

So I expect something like this. You should get closer to the maximum deceleration of 7.5 m/s/s because it is moving to and from that faster.
[image]

Forgive the quick rough sketch.

> If I turned off input shaping (ATC_RATE_FF_ENABLE) would I get the desired braking effect do you think?

No, this won't change anything. This is only the way the attitude controller moves. It still follows the attitude command from the position controller.

GalacticGreg, 2024-03-18:
I changed PSC_JERK_XY to 30 (and upgraded to 4.5beta3) and the brake feature works great now. Thanks a lot for helping me with this!

I feel that the brake mode should have this automatically set to something high, but I'll leave that to you to decide if you want to close this issue.

I bought a 32gb Lexar C10 card, freshly formatted it, and I'm still getting bad logging errors BTW. I'll explore this more later.

lthall, 2024-03-18:
Thanks for the update. I will close this issue.

closed 2024-03-18 by lthall

GalacticGreg, 2024-03-18:
A final comment for those looking into SD card issues. BRD_SD_SLOWDOWN=2 solved my logging problems.

rmackay9, 2024-03-19:
We could add more information to this wiki page to clarify what parameter changes would make it stop faster.
https://ardupilot.org/copter/docs/brake-mode.html#brake-mode

lthall, 2024-03-19:
@GalacticGreg I didn't know about that parameter. Thanks!!

@rmackay9 There is probably some room to improve the documentation and maybe increase the default.

GalacticGreg, 2024-03-19:
PR for documentation submitted. I left out the 7.5m/s^2 max in the docs to make room for changing this.