ArduPilot/ardupilot #26431: BUG: AP_AHRS_DCM.cpp has a bug at correct_centrifugal in drift_correction function

luweiagi, 2024-03-07:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_AHRS/AP_AHRS_DCM.cpp#L785
```
    if (should_correct_centrifugal() && (_have_gps_lock || fly_forward)) {
        const float v_scale = gps_gain.get() * ra_scale;
        const Vector3f vdelta = (velocity - _last_velocity) * v_scale;
        GA_e += vdelta;
        GA_e.normalize();
        if (GA_e.is_inf()) {
            // wait for some non-zero acceleration information
            _last_failure_ms = AP_HAL::millis();
            return;
        }
        using_gps_corrections = true;
    }
```

rmackay9, 2024-03-07:
@luweiagi,

This is more for @tridge and @priseborough I think.

peterbarker, 2024-11-24:
@luweiagi Isn't that still our best estimate for our velocity?  _last_airspeed might still be valid from synthesis?

If we do need this - might be better to set a flag to indicate the velocity can't be trusted - too much copied logic creeping in here... so a `velocity_good_for_drift_correction` boolean?