ArduPilot/ardupilot #26458: Arduplane 4.4.4 - Buidling fw revo-mini-sd fails

nostromo123, 2024-03-08:
## Bug report

Nearly two years ago I posted about **RevoMini with I2C and SD card supported**
in the forum. 

https://discuss.ardupilot.org/t/revomini-with-i2c-and-sd-card-supported/89676

Now I found, that there is a new request by another user for advice about the SD card support.

As I did my 'experiments' mainly with AP 4.1.7, I thought it to be a better idea to try it with AP 4.4.4

My workstation is Debian 12 bookworm, but there seem to be issues with having a building environment on that version, so I installed Debian 11 bullseye in a VM.

No problems with AP 4.4.4 building the fw for **revo-mini** and **revo-mini-i2c**. 

But with **revo-mini-sd** on 4.4.4 the command `~/ardupilot$ ./waf plane`
stops unfinished with

```
.
.
.
[821/826] Compiling ArduPlane/qautotune.cpp
[822/826] Linking build/revo-mini-sd/bin/arduplane
**/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld:common.ld:202 cannot move location counter backwards (from 0000000008103080 to 0000000008100000)**
collect2: error: ld returned 1 exit status

Waf: Leaving directory `/home/kalle/ardupilot/build/revo-mini-sd'
Build failed
 -> task in 'bin/arduplane' failed (exit status 1): 
	{task 140243501086304: cxxprogram AP_Arming.cpp.49.o,ArduPlane.cpp.49.o,Attitude.cpp.49.o,GCS_Mavlink.cpp.49.o,GCS_Plane.cpp.49.o,Log.cpp.49.o,Parameters.cpp.49.o,Plane.cpp.49.o,RC_Channel.cpp.49.o,afs_plane.cpp.49.o,altitude.cpp.49.o,avoidance_adsb.cpp.49.o,commands.cpp.49.o,commands_logic.cpp.49.o,control_modes.cpp.49.o,ekf_check.cpp.49.o,events.cpp.49.o,failsafe.cpp.49.o,fence.cpp.49.o,is_flying.cpp.49.o,mode.cpp.49.o,mode_LoiterAltQLand.cpp.49.o,mode_acro.cpp.49.o,mode_auto.cpp.49.o,mode_autotune.cpp.49.o,mode_avoidADSB.cpp.49.o,mode_circle.cpp.49.o,mode_cruise.cpp.49.o,mode_fbwa.cpp.49.o,mode_fbwb.cpp.49.o,mode_guided.cpp.49.o,mode_loiter.cpp.49.o,mode_manual.cpp.49.o,mode_qacro.cpp.49.o,mode_qautotune.cpp.49.o,mode_qhover.cpp.49.o,mode_qland.cpp.49.o,mode_qloiter.cpp.49.o,mode_qrtl.cpp.49.o,mode_qstabilize.cpp.49.o,mode_rtl.cpp.49.o,mode_stabilize.cpp.49.o,mode_takeoff.cpp.49.o,mode_thermal.cpp.49.o,mode_training.cpp.49.o,motor_test.cpp.49.o,navigation.cpp.49.o,parachute.cpp.49.o,qautotune.cpp.49.o,quadplane.cpp.49.o,radio.cpp.49.o,reverse_thrust.cpp.49.o,sensors.cpp.49.o,servos.cpp.49.o,soaring.cpp.49.o,system.cpp.49.o,tailsitter.cpp.49.o,takeoff.cpp.49.o,tiltrotor.cpp.49.o,tuning.cpp.49.o -> arduplane}
 (run with -v to display more information)
```

To do a cross check I built **revo-mini-sd** with AP 4.3.1 without a problem. (I knew, that in 4.3.1 it was working before) Tested that fw successfully here.
 
I attach both terminal outputs **4.3.1 i2c_sd working** and **4.4.4 i2c_sd failed**

beginning with (deleted the ~/ardupilot folder before)

`~$ git clone --recurse-submodules https://github.com/ArduPilot/ardupilot.git`

[4.3.1 i2c_sd working.txt](https://github.com/ArduPilot/ardupilot/files/14538800/4.3.1.i2c_sd.working.txt)
[4.4.4 i2c_sd failed.txt](https://github.com/ArduPilot/ardupilot/files/14538801/4.4.4.i2c_sd.failed.txt)

All I could find, the building environment might be trying to build a fw, which could be a little too large for the revo-mini.

br KH

IamPete1, 2024-03-08:
"cannot move location counter backwards" means it ran out of flash. You will have to find some other features to disable.

nostromo123, 2024-03-08:
Any suggestion please for modifying the **hwdef.dat** files in ~/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/revo-mini/, ... /revo-mini-i2c/ or .../revo-mini-sd/?

I have to admit, I don't have the slightest idea, what features to disable ....
Some help would be greatly appreciated. I would do the modifications here  and report back.

Regards KH

br KH

IamPete1, 2024-03-08:
The easy one is just to turn off quadplane by adding `define HAL_QUADPLANE_ENABLED 0` to the hwdef.

By the way, you can get the build from the custom build server too. It makes it much easyer to see what options you have for turning stuff off.

[image]

nostromo123, 2024-03-08:
@IamPete1.

That seems helpful. 
`define HAL_QUADPLANE_ENABLED 0` in **revo-mini-sd** hwdef I guess, not in **revo-mini-i2c**!?

I'll test and come back here .....

I'll also have a look into 'Custom Firmware Builder'.

For your information. I made some test and it showed, that this issue here come up when **4.3.3** was released. **4.3.2** is still ok.

For my background, I am a retired mechanical and marine engineer, who likes to fiddle around with things like that, but the last time I did programming myself, was with FORTRAN IV on a PDP11/10. And that is a very very long time ago.

br KH

nostromo123, 2024-03-08:
@Peter.

That single line in did it, building worked!  Simple like that for those who know.
[4.4.4 i2c_sd working.txt](https://github.com/ArduPilot/ardupilot/files/14542899/4.4.4.i2c_sd.working.txt)
Flashed it on the board and made a short test. OLED on I2C ok and LOGs recorded.

Would be nice, if that mod could be added in the next stable. The revo-mini might be a bit old, but still a nice compact FC for planes ..... Could be, that some of those are still around ;-)

br Karl

IamPete1, 2024-03-09:
The revo min normal builds OK and is tested as part of the release, its just the SD variant that we thought was a bit too unusual to build by default, so it got broken. That no one noticed until now says we were probably OK to not test it. In any case, I think we got the bottom of this issue, so I'm going to close.

closed 2024-03-09 by IamPete1