ArduPilot/ardupilot #26478: Scripting restart sometimes stops scripting

peterbarker, 2024-03-11:
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

Sometimes issuing a scripting restart scripting stops instead of restarting

**Version**
master

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
n/a

**Hardware type**
sitl

**Logs**

https://productionresultssa6.blob.core.windows.net/actions-results/62671f2a-c2d1-4f2c-9bf5-6369c1d3a9d3/workflow-job-run-d03d3150-d197-5ecc-2070-a42b9b69a29b/logs/job/job-logs.txt?rsct=text%2Fplain&se=2024-03-10T23%3A44%3A05Z&sig=FFKgLHFL6GBYoN6Yqeh7rhM8x3%2B%2FiKyjjoSE3I7WaeQ%3D&sp=r&spr=https&sr=b&st=2024-03-10T23%3A34%3A00Z&sv=2021-12-02

```
2024-03-10T22:36:39.5820414Z AT-0501.7: Restarting Scripting
2024-03-10T22:36:39.5821166Z AT-0501.7: Sending COMMAND_INT to (1,1) (MAV_CMD_SCRIPTING=42701) (p1=3.000000 p2=0.000000 p3=0.000000 p4=0.000000 p5=0 p6=0  p7=0.000000)
2024-03-10T22:36:39.5822372Z AT-0501.7: ACK received: COMMAND_ACK {command : 42701, result : 0, progress : 0, result_param2 : 0, target_system : 250, target_component : 250} (0.000000s)
2024-03-10T22:36:39.5823211Z AT-0501.7: Waiting for text : shiplanding: loaded
2024-03-10T22:36:39.5823738Z AT-0501.7:   statustext got=(Have beacon) want=(ShipLanding: loaded)
2024-03-10T22:36:39.5824206Z AT-0501.7: AP: Scripting: stopped
2024-03-10T22:36:39.5824542Z AT-0501.8: AP: Airspeed 1 calibrated
```

IamPete1, 2024-03-12:
We should always get "Scripting: stopped" in the restart case, but it should always then restart.

IamPete1, 2024-03-12:
Maybe we just need to wait longer in CI for the next message to come through?

tpwrules, 2025-09-27:
Example A:

sitl log: [QuadPlane-PrecisionLanding.txt](https://github.com/user-attachments/files/22575464/QuadPlane-PrecisionLanding.txt)

```
2025-09-25T16:48:20.2807590Z AT-0571.0: Restarting Scripting
2025-09-25T16:48:20.2808035Z AT-0571.0: Sending COMMAND_INT to (1,1) (MAV_CMD_SCRIPTING=42701) (p1=3.000000 p2=0.000000 p3=0.000000 p4=0.000000 p5=0 p6=0  p7=0.000000 f=5)
2025-09-25T16:48:20.2808788Z AT-0571.0: ACK received: COMMAND_ACK {command : 42701, result : 0, progress : 0, result_param2 : 0, target_system : 250, target_component : 250} (0.000000s)
2025-09-25T16:48:20.2809326Z AT-0571.0: Waiting for text : plnd: loaded
2025-09-25T16:48:20.2809595Z AT-0571.0: AP: Scripting: stopped
2025-09-25T16:48:20.2809822Z AT-0571.1: AP: Airspeed 1 calibrated
2025-09-25T16:48:20.2810081Z AT-0571.6: AP: GPS 1: probing for u-blox at 230400 baud
2025-09-25T16:48:20.2810360Z AT-0571.6: AP: GPS 1: detected u-blox
```

bin log: [QuadPlane-PrecisionLanding-00000027.zip](https://github.com/user-attachments/files/22575457/QuadPlane-PrecisionLanding-00000027.zip)

```
2025-09-25 11:48:19.037 SRC=250/250:AT-0571.0: Restarting Scripting
2025-09-25 11:48:19.114 SRC=250/250:AT-0571.0: Sending COMMAND_INT to (1,1) (MAV_CMD_S
2025-09-25 11:48:19.114 SRC=250/250:CRIPTING=42701) (p1=3.000000 p2=0.000000 p3=0.0000
2025-09-25 11:48:19.114 SRC=250/250:00 p4=0.000000 p5=0 p6=0  p7=0.000000 f=5)
2025-09-25 11:48:19.127 Scripting: stopped
2025-09-25 11:48:19.127 SRC=250/250:AT-0571.0: ACK received: COMMAND_ACK {command : 42
2025-09-25 11:48:19.127 SRC=250/250:701, result : 0, progress : 0, result_param2 : 0, 
2025-09-25 11:48:19.127 SRC=250/250:target_system : 250, target_component : 250} (0.00
2025-09-25 11:48:19.127 SRC=250/250:0000s)
2025-09-25 11:48:19.127 SRC=250/250:AT-0571.0: Waiting for text : plnd: loaded
2025-09-25 11:48:20.135 Scripting: restarted
2025-09-25 11:48:20.149 PLND: Loaded
2025-09-25 11:48:20.380 Airspeed 1 calibrated
2025-09-25 11:48:26.000 GPS 1: probing for u-blox at 230400 baud
2025-09-25 11:48:26.020 GPS 1: detected u-blox
```

The autopilot generated and logged the messages but they did not make it to the test suite.

tlog excerpt: [QuadPlane-PrecisionLanding-autotest-1758818327286535-lastmeg.zip](https://github.com/user-attachments/files/22575482/QuadPlane-PrecisionLanding-autotest-1758818327286535-lastmeg.zip)

(passed through `strings`)
```
AT-0571.0: Restarting Scripting
8R/?`
-sD{
AT-0571.0: Sending COMMAND_INT to (1,1) (MAV_CMD_S
CRIPTING=42701) (p1=3.000000 p2=0.000000 p3=0.0000
00 p4=0.000000 p5=0 p6=0  p7=0.000000 f=5)
AT-0571.0: ACK received: COMMAND_ACK {command : 42
701, result : 0, progress : 0, result_param2 : 0, 
target_system : 250, target_component : 250} (0.00
0000s)
AT-0571.0: Waiting for text : plnd: loaded
Scripting: stopped
8R/?gH
[...]
;+64;
ILua: State m
emory us
age: 482
4 + 10
PLND: Loaded
Z_#?d_#?
[...]
&;v%
Airspeed 1 calibratedX
ARSPD_OFFSET
 [...]
GPS 1: probing for u-blox at 230400 baud
GPS 1: detected u-blox
```

Looks like the script message got transmitted but probably corrupted by the adjacent "Lua: State memory usage" raw print. Look about 0xA3C88 in the file. I don't see the "restarted" message though which is strange. Interestingly plot.ardupilot.org can see "PLND: Loaded" but the timestamp is bogus.

tpwrules, 2025-09-27:
Example B:

sitl log: [ArduCopter-ScriptParamRegistration.txt](https://github.com/user-attachments/files/22575547/ArduCopter-ScriptParamRegistration.txt)

```
AT-0802.3: Restarting Scripting
AT-0802.3: Sending COMMAND_INT to (1,1) (MAV_CMD_SCRIPTING=42701) (p1=3.000000 p2=0.000000 p3=0.000000 p4=0.000000 p5=0 p6=0  p7=0.000000 f=5)
AT-0802.3: ACK received: COMMAND_ACK {command : 42701, result : 0, progress : 0, result_param2 : 0, target_system : 250, target_component : 250} (0.000000s)
AT-0802.3: Waiting for text : restart
AT-0802.4: AP: test script running
AT-0802.4: AP: Scripting: stopped
AT-0802.4: AP: EKF3 IMU1 tilt alignment complete
AT-0802.4: AP: EKF3 IMU0 tilt alignment complete
AT-0802.4: AP: EKF3 IMU1 MAG0 initial yaw alignment complete
AT-0802.4: AP: EKF3 IMU0 MAG0 initial yaw alignment complete
AT-0802.5: AP: test script running
AT-0802.5: AP: test script running
```

Scripts are clearly running but it's absolutely impossible for them to be after the "stopped" message is printed. So they must have successfully restarted.

bin log: [ArduCopter-ScriptParamRegistration-00000132.zip](https://github.com/user-attachments/files/22575561/ArduCopter-ScriptParamRegistration-00000132.zip)

```
2025-09-27 01:16:58.430 SRC=250/250:AT-0802.3: Restarting Scripting
2025-09-27 01:16:58.652 SRC=250/250:AT-0802.3: Sending COMMAND_INT to (1,1) (MAV_CMD_S
2025-09-27 01:16:58.652 SRC=250/250:CRIPTING=42701) (p1=3.000000 p2=0.000000 p3=0.0000
2025-09-27 01:16:58.652 SRC=250/250:00 p4=0.000000 p5=0 p6=0  p7=0.000000 f=5)
2025-09-27 01:16:58.660 SRC=250/250:AT-0802.3: ACK received: COMMAND_ACK {command : 42
2025-09-27 01:16:58.662 SRC=250/250:701, result : 0, progress : 0, result_param2 : 0, 
2025-09-27 01:16:58.662 SRC=250/250:target_system : 250, target_component : 250} (0.00
2025-09-27 01:16:58.662 SRC=250/250:0000s)
2025-09-27 01:16:58.662 SRC=250/250:AT-0802.3: Waiting for text : restart
2025-09-27 01:16:59.280 test script running
2025-09-27 01:16:59.281 Scripting: stopped
2025-09-27 01:16:59.587 EKF3 IMU1 tilt alignment complete
2025-09-27 01:16:59.592 EKF3 IMU0 tilt alignment complete
2025-09-27 01:16:59.685 EKF3 IMU1 MAG0 initial yaw alignment complete
2025-09-27 01:16:59.690 EKF3 IMU0 MAG0 initial yaw alignment complete
2025-09-27 01:16:59.690 Event: DATA_EKF_YAW_RESET
2025-09-27 01:17:00.293 Scripting: restarted
2025-09-27 01:17:01.317 test script running
2025-09-27 01:17:02.319 test script running
2025-09-27 01:17:03.336 test script running
```

Again the autopilot successfully generated and logged the restart message.

There's no tlog for this one (one was generated for an earlier test, maybe there is a test harness bug?) so I can't confirm there but it seems likely the same issue.

PR #31180 (2025-09-30): AP_Scripting: avoid `DEV_PRINTF` for initial memory usage
This corrupts other MAVLink messages going out at the same time, particularly in CI, causing tests to fail because scripting restart and script initial startup messages are destroyed.

Instead use a MAVLink debug statustext, but only send it if the runtime measurement debug bit is on to avoid an obvious change in behavior and unnecessary appearance of a new message.

Tested on Cube Orange that the message appears at the right time. I don't love that it now requires an extra step and additional messages to see this but the CI fix is impo [... cut]
closed 2025-09-30 by tridge