ArduPilot/ardupilot #26479: AP_Mount: Return to RC_Targeting automatically when control is sent by mavlink instead of Rc channels

Davidsastresas, 2024-03-11:
Support for this under Rc control was added here:

https://github.com/ArduPilot/ardupilot/pull/25927

But if we are controlling the gimbal by mavlink manually, for example using a Joystick in the GCS and gimbal_manager_pitchyaw messages, the PR above doesn't apply.

This implies that we don't have yaw lock/follow control when in mavlink targeting, so if using a joystick or GCS controls, we don't have a way to make Ardupilot behave correctly regarding yaw lock/follow.

rc and mavlink targetting in AP are concepts that are brought historically by the old implementation, but I am not sure they make a lot of sense nowadays. I think they add extra complexity and they only benefit, which would be to know who is in control, if RC or mavlink, is contemplated in the new gimbal manager v2 protocol, in gimbal_device_flags:

```
GIMBAL_DEVICE_FLAGS_RC_EXCLUSIVE=256, /* The gimbal orientation is set exclusively by the RC signals feed to the gimbal's radio control inputs. MAVLink messages for setting the gimbal orientation (GIMBAL_DEVICE_SET_ATTITUDE) are ignored. | */
   GIMBAL_DEVICE_FLAGS_RC_MIXED=512, /* The gimbal orientation is determined by combining/mixing the RC signals feed to the gimbal's radio control inputs and the MAVLink messages for setting the gimbal orientation (GIMBAL_DEVICE_SET_ATTITUDE). How these two controls are combined or mixed is not defined by the protocol but is up to the implementation. | */
   ```
   
   For awareness @rmackay9.

olliw42, 2024-03-11:
argh, do NOT misuse/misinterpret these flags
listen to what they say: **"by the RC signals feed to the gimbal's radio control inputs"** !!! The gimbal's rc input is hardly the gimbal manager aka flight controller in ArduPilot's case.
don't screw up the protocol, better use the protocol's ways for "RC control".

Davidsastresas, 2024-03-11:
Thanks for the clarification @olliw42. I had to actually re read a couple of times the description to get your point. You are right.

I need to rethink and test to understand what I am asking for here then. Closing this for now.

closed 2024-03-11 by Davidsastresas

Davidsastresas, 2024-03-11:
In any case @olliw42 , I have to say, there are many ways to address stuff like this, and I find your tone is really unfortunate for the situation. 

I am really not sure what you have on your mind when you address situations like that, but it would not harm to be nicer. There is no need to address me as if I were dumb and I wanted to mess with the protocol on purpose, it isn't the case, and I think you know that. It would not harm you to re-read such comments twice before publishing them. They are really out of place.

olliw42, 2024-03-11:
There is substantial evidence that you guys do have a talent to screw up a protocol - if on purpose or other reason doesn't matter, the proof is in the taste of the pudding as the saying goes. I can tell you what I have on mind then I read this and other such stuff: "argh". I went very silent on gimbal & camera stuff, here I however felt that the magnitude of potential damage waranted interference. And I feel it worked, that's all what counts in my books. And btw my tone is reaction, not action. As simple as that.

rmackay9, 2024-03-11:
@olliw42,

One of the core principles of getting along with people is not to get personal.  Please watch yourself and how you communicate.   This has been brought up many times.

Davidsastresas, 2024-03-11:
@olliw42 Who do you think you are to speak like that? Did I ever disrespected you for you to speak to me like that? Because from what I recall from my interactions with you I think it was pretty much the opposite, I think I always addressed you with respect, and I have listened to your advice every single time you have spoken to me.

A year ago you manifested you disagreed with how we were planning all this gimbal manager implementation in QGC, and I respected your opinion as much as for listening to your thoughts in a call for a couple hours. And I have manifested several times here and in QGC repo that I am all in with bringing this protocol to life. Since then, together with other devs like Randy, Julian, Hamish, etc we have spent many time fine tuning the protocol, and implementing it in all ends.

Because a year back, the protocol wasn't ready. There were some fundamental details that needed to be fixed for it to work in the real world. And back then you were not aware of it.

Did you intervene all this time? Have you been around when we were discussing and adjusting the protocol so it gets to the point where we can all use it effectively? No, you were not. So I don't understand why you think you are so entitled to threat us with such disrespect when you haven't even been involved in this last year development's on this matter.

So now I open an issue, not even a PR, and I make a wrong side statement because I miss-read a comment, and you go all "talent to screw up a protocol" "taste of the pudding" "argh", and your tone is "reaction".

"And I feel it worked, that's all what counts in my books". Do you realize you could have made it work without being an asshole? Like I said above, I have listened to you and applied your suggestions many times in the past, and you know it. **If that is really only what counts in your books you really need to check your ego and your perception of reality because both of them are pretty screwed up.** 

This is the last time I will listen or answer to you if you keep disrespecting.

Don't even bother to answer if it isn't to apologize, I won't follow this anymore.