ArduPilot/ardupilot #26481: Plane: Disable motors when altitude about zero with lidar

hyperion11, 2024-03-11:
## Feature request

If you land the plane away from you and do not have time to turn on the stop motor, the RTL may be triggered, the motors will start and the plane may be damaged.

I propose to add a setting to turn off motors when there is a lidar. We check the alt relative to the home so that it is <= 0. We check the alt  using the lidar. If both conditions are met, turn off the motors. I'm thinking about adding airspeed to the check. If it is below FBW_MIN (or less than 2-3 m\s), then turn it off, if more, then not until it becomes less

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2024-03-11:
Throttle will be suppressed if landing is detected.

hyperion11, 2024-03-13:
> Throttle will be suppressed if landing is detected.
in my case it didn't happen

LupusTheCanine, 2024-05-21:
> > Throttle will be suppressed if landing is detected.
> in my case it didn't happen

Landing can be detected only if you are using landing feature IIRC, then motor stop is a part of the flare. Otherwise it is called crash check and can only act after hitting the ground hard enough.

magicrub, 2024-05-22:
not all rangefinders are created equal. Some have the nasty tendency to report 0 when not on the ground due to moister or other environmental effects.

I think what you're looking for is the CRASH_DETECT param which will do what you want it to do but without a lidar dependency

geofrancis, 2024-05-22:
what about something like a micro switch on the belly? you could set it up as a button to trigger an aux function or lua script.

LupusTheCanine, 2024-05-22:
> not all rangefinders are created equal. Some have the nasty tendency to report 0 when not on the ground due to moister or other environmental effects.
> 
> I think what you're looking for is the CRASH_DETECT param which will do what you want it to do but without a lidar dependency

Yup, I am testing LD2450 (24Ghz human presence sensor that can work as a rangefinder) and it returns 0s for no detections, thankfully it includes a 0 in a field that is consistently not 0 when target is detected but still.