ArduPilot/ardupilot #26492: Servo output based on arming state. (like servo 30 function)

VinceHogg, 2024-03-12:
Servo function 30 is described as 'engine run enable' or 'motor enable' and is supposed to output 2 PWM values depending on arming state. It would also be useful for turning on or off equipment before flight. However in copter 4.4.4 it seems to do nothing. Perhaps it works when ICE_ENABLE=1 ? This parameter in not in copter.
Could it be enabled for all vehicles as a general propose feature?

Thanks.

IamPete1, 2024-03-12:
Run enable is a helicopter feature. 

It would be a quite simple lua script to do as you describe. Something like:

```
-- use scripting 1 servo output
local SRV_function = 94

-- set scaled range as 0 to 1
SRV_Channels:set_range(SRV_function, 1.0)

function update()
  if arming:is_armed() then
    SRV_Channels:set_output_scaled(SRV_function, 1.0)
  else
    SRV_Channels:set_output_scaled(SRV_function, 0.0)
  end

  return update, 100 -- run at 10 Hz
end

return update() 
```

VinceHogg, 2024-03-12:
Thanks for that. It may be simple but its been 20 years since I did any programming. I should get my self more familiar with it. 
It may be a good idea to open that servo 30 feature for all vehicles and perhaps rename. It could still be used for the original purpose.

peterbarker, 2024-07-31:
> Thanks for that. It may be simple but its been 20 years since I did any programming. I should get my self more familiar with it. It may be a good idea to open that servo 30 feature for all vehicles and perhaps rename. It could still be used for the original purpose.

I don't think re-purposing 30 is a good idea; it might be that the user wants to remain armed while also killing the motor.  For example, we have an option to turn on the safety switch when you disarm - a user might want their ailerons to continue to function even when they've chosen to kill the motor in-flight.

@VinceHogg is @IamPete1 's script enough to get you by?

peterbarker, 2025-05-11:
OK, no response from @VinceHogg , and the script saves us maybe 64 bytes of flash ;-)

Closing this one now

closed 2025-05-11 by peterbarker