ArduPilot/ardupilot #26519: Copter: Auto mode Waypoint commands may complete even if the vehicle never reaches the WP

rmackay9, 2024-03-15:
Copter's Auto mode may eventually consider a waypoint command complete even though the vehicle never gets close to the waypoint location.  Below is a screen shot of a SITL test in which the vehicle is attempting to fly a square mission but the simulated wind has been set to blow strongly towards the North.  The vehicles reaches WP2 but is unable to reach WP.  Despite this eventually WP3 is considered complete and the vehicle moves onto WP4.
[image]

This can be recreated in SITL by doing the following

- start SITL and enter a square mission similar to the above screenshot.  Ensure there are no delays at the waypoints.
    
  [image]

- param set SIM_WIND_SPD 10
- param set ANGLE_MAX 1000
- arm and takeoff and fly the mission

I think the issue is [this line in AC_WPNav](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_WPNav/AC_WPNav.cpp#L485) which does not allow the "Spline time" to drop below 0.1.

lthall, 2024-03-15:
Yes, I think this is the case for a fast waypoint.

https://github.com/ArduPilot/ardupilot/blob/b316b3ab46074d2499c0143cd2fdc7c69dc5a8a9/libraries/AC_WPNav/AC_WPNav.cpp#L547C47-L547C56

The check above doesn't happen on a fast waypoint but does on a slow waypoint. We could run that check on both. It shouldn't make a difference for normal situations but if the aircraft is getting blown off it may not be quiet as smooth. That is probably ok though.

xianglunkai, 2024-03-18:
@rmackay9 
May I ask how to replicate the above situation on the boat using MissionPlanner？
Txs!

rmackay9, 2024-03-19:
@xianglunkai,

Sure, just set the SIM_TIDE_SPEED parameter to a high number like 9.  By the way, if you make it too high so that the boat/rover actually moves backwards an EKF failsafe may be triggered which causes the vehicle to switch to Hold mode.  This is a separate issue caused by the EKF's assumption that boats always move forward.  We may need a setting to disable this assumption.. but this is a separate issue.

xianglunkai, 2024-03-19:
@rmackay9 
Thank you very much!

lthall, 2025-05-09:
Ok, the first thing is we are looking at an aircraft that is unable to control it's position, so the only thing we are looking at here is that the fact that fast waypoints don't currenlty check the instantainous distance to the waypoint to declare we have reached the waypoint.

Fast waypoints are treated as just one point on a kinimatic path. As such the WP_Nav libary manages the position along the kinimatic path, through fast waypoints, to ensure the target position doesn't move further along the path than the aircraft is currently positioned. This is done correctly here.

This is confusing because while the aircraft position is not under control, it is drifting in the general direction of the specified path. As such the instantainouos target along that path is moved only slightly in front of the minimum cross track distance. 

This will continue until the direction of the path, and therefore the target location, moves away from the current aircraft location. At this point the speed along the track will decay exponentially towards zero. As this is an exponential decay it won't ever reach zero so we will still move towards the next waypoint, but at an ever slower rate. This is the change made in #26520, previously it could only slow down to 10%.

I wouldn't call this a bug as the code is working exactly as we intend. Instead I would ask some questions:
- Do we want to slow down the progression along the path if the cross track error becomes large compared to the waypoint radius. 
- Should we change the slowing along the path to a liniar rather than exponential decay so we do come to a complete stop.
- Should we abort a mission and call some sort of failsafe if we can't control position past some threashold. (especially as this is required to demonstrate this behaviour)