ArduPilot/ardupilot #26540: sim_vehicle.py not sending data to FlightGear

ugol-1, 2024-03-16:
## Bug report

**Issue details**

Using FlightGear with ArduPilot SITL as described in https://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#flightgear-3d-view-optional does not work. 

Steps to reproduce:
1. Set up ArduPilot build environment as described in https://ardupilot.org/dev/docs/building-setup-linux.html
2. Install FlightGear:
    ```bash
    sudo apt install flightgear
    ```
3. Run FlightGear:
    ```bash
    cd ardupilot && Tools/autotest/fg_plane_view.sh
    ```
4. In a second console, run ArduPlane SITL:
    ```bash
    sim_vehicle.py -N -v ArduPlane
    ```
5. Load mission and take-off:
    ```
    wp load Tools/autotest/Generic_Missions/CMAC-circuit.txt
    mode auto
    arm throttle
    ``` 

Expected behavior: the aircraft starts moving in FlightGear
Actual behavior: the aircraft does not move in FlightGear
    
**Version**
Ardupilot: https://github.com/ArduPilot/ardupilot/commit/5cd18dbc451e91d65d686cad5d57ebd343171abc
FlightGear version: 2020.3.18

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
any

**Hardware type**
SITL

**Logs**
MAVProxy console:
```
wp load Tools/autotest/Generic_Missions/CMAC-circuit.txt
MANUAL> Loaded 8 waypoints from Tools/autotest/Generic_Missions/CMAC-circuit.txt
Loaded 8 waypoints in 0.15s
Sent all 8 waypoints
Got MISSION_ACK: TYPE_MISSION: ACCEPTED
AP: Flight plan received
mode auto
MANUAL> Got COMMAND_ACK: DO_SET_MODE: ACCEPTED
AUTO> Mode AUTO
AP: Mission: 1 Takeoff
200
waypoint 1
arm throttle
AUTO> Got COMMAND_ACK: COMPONENT_ARM_DISARM: ACCEPTED
AP: Throttle armed
AP: Armed AUTO, xaccel = -0.0 m/s/s, waiting 0.2 sec
AP: Triggered AUTO. GPS speed = 0.0
ARMED
AP: Holding course 35279 at 17.7m/s (-0.0)
height 15
height 26
AP: Takeoff level-off starting at 17m
height 37
AP: Takeoff complete at 41.45m
AP: Mission: 2 WP
height 47
400
waypoint 2
200
```

FlightGear window:
[image]

ugol-1, 2024-03-16:
Sending data to FlightGear was disabled by default in 747afd7673e18ab2a8b324fe8ec6c3a9722cea4a, this is why it stopped working.

peterbarker, 2024-11-24:
I believe this issue was addressed in #27993 

Can you confirm, @ugol-1 ?

peterbarker, 2025-05-11:
Thanks for following up, @ugol-1 - closing this one now.

closed 2025-05-11 by peterbarker