ArduPilot/ardupilot #26553: EKF3 IMU0/1 forced reset

CasADi-acados, 2024-03-17:
## Bug report
_Please describe the problem_
[image]
The heading angle output from my EKF3 always gets reset and cleared to zero. I've checked the logs and still can't see the problem. Can anyone please help me? My work log is as follows.
Some people have analyzed that it may be a problem with the accelerometer and needs to be recalibrated, but I feel that my accelerometer data is correct.

**Logs**
_Please provide a link to any relevant logs that show the issue_
[image]
[image]
[image]
[image]
[2024-03-17 16-53-58.zip](https://github.com/ArduPilot/ardupilot/files/14627808/2024-03-17.16-53-58.zip)

CasADi-acados, 2024-03-17:
[image]
Figure 1 shows the original data of imu and the data after data fusion, which was not uploaded successfully before.

CasADi-acados, 2024-03-17:
[image]
This is the accelerometer data, which I also think is fine.

rmackay9, 2024-03-18:
Hi @CasADi-acados,

Thanks for the report.  Support issues really should go into the support forums but I'll leave it here because we don't do much sub support on the AP discord servers. That's normally handled by BlueRobotics but I see in this case that you're not using the BlueRobotics hardware (which is a-ok of course).

Any chance you could use a more modern version of AP?  4.1 is quite old.

CasADi-acados, 2024-03-18:
> Hi @CasADi-acados,
>
> Thanks for the report. Support issues really should go into the support forums but I'll leave it here because we don't do much sub support on the AP discord servers. That's [...]

Thank you very much for your reply. I currently think it is a problem with the EKF3 filter algorithm. I don't know which data abnormality caused its forced clearing.

amilcarlucas, 2024-03-19:
@CasADi-acados please answer the question:

Any chance you could use a more modern version of AP?

CasADi-acados, 2024-03-20:
> @CasADi-acados please answer the question:
> 
> Any chance you could use a more modern version of AP?
Is there a newer version of ArduSub? I haven't found it yet.

CasADi-acados, 2024-03-20:
> @CasADi-acados please answer the question:
> 
> Any chance you could use a more modern version of AP?

[image]
May I ask, are these the latest ones?

CasADi-acados, 2024-03-20:
> Hi @CasADi-acados,
>
> Thanks for the report. Support issues really should go into the support forums but I'll leave it here because we don't do much sub support on the AP discord servers. That's [...]

[image]
May I ask, are these the latest ones?

CasADi-acados, 2024-03-20:
I think it's a problem with EKF3. I don't know why there are burrs in the curves of my two graphs. I feel like my acceleration curves are quite normal.

[image]
[image]