ArduPilot/ardupilot #26555: Tracker: support autopilot mounted on non-moving portion of tracker frame

rmackay9, 2024-03-18:
Tracker's current implementation requires that the autopilot be mounted on the moving portion of the frame.  This is required so that the autopilot's IMU can be used to measure the attitude of the antenna directly.

There is a [request here in the forums](https://discuss.ardupilot.org/t/fc-fixed-in-frame/114827) to allow mounting the autopilot on a non-moving part of the frame.

This is actually quite a significant change so I'm not sure if we'd actually want to implement this.

jorgeneo560, 2024-03-18:
thanks for taking in consideration the idea, i could help to modify the code and let the user decide wich way to go.
Add me to the next developer talk so they could explain me what parts of the code should modify to add this funcionality, i am having some issues with the mini 880 gps+compass and the holybro 915mhz 500mw modem, when i switch to 100mw modems compass mag field warning dissapear, so having the compass far away from the antenna and modem it would be nice, and having the fc fixed in place it would be a great help.

amilcarlucas, 2024-03-18:
To preserve the accuracy it should remain a closed loop control system, meaning it needs sensors to measure the pitch and yaw of the antenna. How do you propose to include those? and how to calibrate them?

jorgeneo560, 2024-03-18:
theres two solutions here at my point of view, a servo is a closed loop system it self, if we set the degree angle in a pwm range when the tracker wants to pitch it should set the pwm value and assume its ok, if the user is using let say a 360 servo or a brushless motor or just wanted to had closed loop also a feedback is needed, for this we could attach a external imu sensor to the antenna and use it in the i2c bus that dosent require too much wiring. We could give the user the option for going to open loop for pitch with a warning in mission planner or give the user the option for a external imu.

One of the weak points of the ardupilot antenna tracker is the need of the fc in the pitch structure, this make wiring complex and bulky structures, if we make the modification for allow to leave the fc fixed in the base frame it would be posible to use ardupilot firmware on manny populars antenna trackers frames.

Also having the fc away from the antenna it would increase compass reliability.

rmackay9, 2024-03-19:
@jorgeneo560,

Thanks for the feedback.  In general I think an open loop is possible although its a large change.  It's so large that it might not be possible to find a dev to put in the effort but let's see.

Re the external IMU, some autopilots are so tiny these days that I think using one of these would be a simpler alternative.

Anyway, we don't actually all have to agree at this point.  We only need to agree 5min before a developer actually starts work on the idea.

jorgeneo560, 2024-03-19:
i could do the modification, i just need a quick review of the code to start working on it.