ArduPilot/ardupilot #26557: Copter navigation without compass

geofrancis, 2024-03-18:
iNAV has recently introduced a feature in 7.1 that lets copters navigate without a compass, I think this would be a good idea for a backup heading for ardupilot if compass readings become unreliable since currently the backup is just althold.

https://github.com/iNavFlight/inav/pull/9387

https://www.youtube.com/watch?v=iopZfH-DdTI

closed 2024-03-18 by geofrancis

rmackay9, 2024-03-19:
Yes, GSF should allow flying without a compass (if a GPS is present) although we also have an outstanding issue with GSF that it may not work unless the vehicle is flying at 5m/s.  We've already got a PR to reduce that though and I plan to get it into 4.5. https://github.com/ArduPilot/ardupilot/pull/24206

geofrancis, 2024-03-19:
Inav recently added a lot of stuff that might be worth looking at 

https://github.com/iNavFlight/inav/pull/9387