ArduPilot/ardupilot #26617: AP_Torqeedo: add TorqLink support

rmackay9, 2024-03-25:
The higher powered Torqeedo motors use TorqLink which we think is J1939 CAN.

Relevant links are here:

- https://discuss.ardupilot.org/t/torqeedo-cruise-3-0-fp-fixed-pod-compatibility/104490
- a community developer, Nubbish, wrote a Lua script that apparently works.  I have put a copy of it here: https://www.dropbox.com/scl/fi/azgidbnmw3a3fsptrmvm2/CAN_TorqLink.lua?rlkey=liw0un5e1q3kjnvdo2x7976sc&dl=0
- [Torqeedo multi support issue](https://github.com/ArduPilot/ardupilot/issues/24216)

rmackay9, 2025-01-17:
For those that are interested in this, I've created a modified version of Nubbish's script.  I'd very much appreciate feedback from someone with the hardware to confirm it still works before I move onto adding support for more feedback from the motor

https://github.com/rmackay9/rmackay9-ardupilot/blob/torqeedo-torqlink-lua/libraries/AP_Scripting/drivers/torqeedo-torqlink.lua

Shift-Ronan, 2025-01-20:
@rmackay9 Nubbish here. I got to testing on our setup this afternoon, and there are a few hiccups.

1. The script threw errors when packing throttle values into `speed_frame`  on line 153 because `speed_cmd` is not an integer. I used `math.round()` to work around that.
2. More errors from this debug print on line 204: `gcs:send_text(MAV_SEVERITY.INFO, string.format("%s read id:%d PGN:%s", TEXT_PREFIX_STR, frame:id(), pgn))`. Solved by using `:toint()` on `frame:id()`  because `frame:id()` returns a `uint32_t`.
3. Motors running! (even if only left side ready state is checked). They seem to timeout every few seconds, even after increasing `RECEIVE_TIMEOUT_MS` up to 5000 (5 seconds).
Video of this issue: [IMG_1212.mov](https://cdn.discordapp.com/attachments/466737881168478228/1331016409731567707/IMG_1212.mov?ex=67901521&is=678ec3a1&hm=80567dcf84968aeac04fbe7a5e0df3288915545f2254e8688cb0fae4a0802c46&)

If you have some ideas for point 3, I can give them a try.

NandoTTU, 2025-02-07:
Hello I recently acquired a torqlink torqeedo motor, is there a wiring pin out diagram showing how to connect?

rmackay9, 2025-02-07:
Hi @NandoTTU,

If you're using the smaller travel then the regular [AP Torqeedo wiki page](https://ardupilot.org/rover/docs/common-torqeedo.html) has a detailed wiring diagram.  For the motors using TorqLink I'm afraid that I don't have one so I really don't know how to connect them.  TorqLink is just NMEA 2000 so it's a serial connection but I don't know the voltage that the Torqeedo motor uses for this so there might need to be an adapter between the autopilot and motor.

Shift-Ronan, 2025-02-22:
> Hello I recently acquired a torqlink torqeedo motor, is there a wiring pin out diagram showing how to connect?

The current setup requires a "J1939 Gateway for 3rd party throttle" from Torqeedo (Part #:  000-00913) which translates the proprietary Torqeedo motor protocol into NMEA 2000/J1939 standard.
The physical output of the gateway follows the NMEA 2000 standard (image below).
[image]

If you want to review the discussion on Discord for more details, see [this thread](https://discord.com/channels/674039678562861068/1280785432707338281).