ArduPilot/ardupilot #26645: Sub: underwater disarm sets baro field elevation to origin.alt

clydemcqueen, 2024-03-29:
## Bug report

**Issue details**

1. set ekf origin using SET_GPS_GLOBAL_ORIGIN or some other method
2. arm
3. dive to any depth
4. disarm underwater

baro._field_elevation_active is set to origin.alt.

If origin.alt is 0 (typical), the field elevation is set to 0 every second as long as the sub is disarmed.

At this point the sub has no idea where it is in the water column.

It looks like this was introduced in https://github.com/ArduPilot/ardupilot/pull/20867

**Version**

master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[ x ] Submarine

**Airframe type**

All

**Hardware type**

All

**Logs**

This is a [lightly edited] SITL run. I'm using a branch with ORIGIN_* parameters that set the EKF origin when the sub is armed, but the same holds true for master with the origin set via SET_GPS_GLOBAL_ORIGIN:

MANUAL> param show ORIGIN*
ORIGIN_ALT       0.0
ORIGIN_LAT       48.07746124267578
ORIGIN_LON       -122.88683319091797
MANUAL> arm throttle
Got COMMAND_ACK: COMPONENT_ARM_DISARM: ACCEPTED
AP: Arming motors
AP: EKF3 IMU0 origin set
AP: EKF3 IMU1 origin set
AP: Using backup location
ARMED
Arming checks disabled
Flight battery 100 percent
MANUAL> rc 3 1200
height -5
MANUAL> disarm
Got COMMAND_ACK: COMPONENT_ARM_DISARM: ACCEPTED
AP: Disarming motors
AP: Field Elevation Set: 0m
height 5
DISARMED
AP: Field Elevation Set: 0m
AP: Field Elevation Set: 0m
AP: Field Elevation Set: 0m
AP: Field Elevation Set: 0m
...

rmackay9, 2024-03-31:
Hi @clydemcqueen,

Could you clarify what the issue is with having the field elevation set while disarmed?  Is it just the message is annoying?  Does it cause problems in some way?

clydemcqueen, 2024-04-01:
Hi @rmackay9 -- it re-calibrates the barometer so that the barometer reports origin.alt (0 in this case) at the current pressure. So the sub thinks it has teleported to the surface.

rmackay9, 2024-04-01:
@clydemcqueen,

Ok, thanks for the clarification.  So this would only happen if the user disarmed while deep underwater though wouldn't it?

clydemcqueen, 2024-04-01:
Yes, it is only a problem for sub if the user disarms while submerged. The sub still operates fine, but the user doesn't know how deep it is.

Williangalvani, 2024-04-01:
I want to point out that is pretty common for me to disarm the rov while submerged. Most of them are slightly buoyant, so it usually just means they will slowly surface.

clydemcqueen, 2024-04-27:
Fixed via https://github.com/ArduPilot/ardupilot/pull/26658

closed 2024-04-27 by clydemcqueen