ArduPilot/ardupilot #26650: Copter: No wind estimate when using EK3_SRC1_YAW = GPS

TheRealKaufmann, 2024-03-29:
## Bug report

**Issue details**
I am using a HERE 4 Dual GPS Configuration for Yaw and it works really well. However, when selecting EK3_SRC1_YAW = (GPS or GPS with Compass Fallback), there are no wind estimates shown in he wind_vel and wind_dir messages. Also, there is no wind estimate shown in the Log under VWN and VWE. The EK3_DRAG_BCOEF_* and EK3_DRAG_MCOEF parameters have been configures correctly. When selecting EK3_SRC1_YAW = Compass, the wind estimates are shown and available. 

**Version**
ArduCopter 4.4.4

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QUAD X

**Hardware type**
Cube Orange

**Logs**
-

rmackay9, 2024-03-30:
Thanks for the report, I've added this to the [Copter-4.5.0 issues list](https://github.com/ArduPilot/ardupilot/issues/26103) although I see you said the same issue appears with 4.4.

In any case, I'll try to reproduce the issue in SITL.

TheRealKaufmann, 2024-03-30:
@rmackay9 Thank you! Could you please give me a hint where in the code this problem might arise? 
I might be able to work it through, if it is only in the EKF. For us, this is a fairly critical thing since we want to take off on a large ship with strong winds soon.

rmackay9, 2024-04-03:
This issue can be reproduced in SITL meaning that it is likely caused by a bug in the EKF.  The steps to reproduce are below:

First confirm that wind estimates work with all default parameters

1. cd ardupilot/ArduCopter
2. ../Tools/autotest/sim_vehicle.py --map --console
3. param load [../Tools/autotest/default_params/copter-wind-estimates.parm](https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/default_params/copter-wind-estimates.parm)
4. restart SITL
5. arm throttle
6. GUIDED
7. takeoff 20
8. status WIND (to show the direction and speed, note the non-zero values)

Next load the gps-for-yaw parameters and repeat the flight
10. param load [../Tools/autotest/default_params/copter-gps-for-yaw.parm](https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/default_params/copter-gps-for-yaw.parm)
11. repeat steps 5, 6, 7 and 8 above and note that the WIND direction is always -180 and speed is 0
FYI @tridge, @priseborough

rmackay9, 2024-07-24:
@TheRealKaufmann,

Thanks very much for the analysis.  I've slightly fixed the formatting of your comment and I think you meant "recordYawResetsCompleted" instead of "recordYawReset()".

Yes, surely this is related to the "canEstimateWind" check [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF3/AP_NavEKF3_Control.cpp#L62).
```
    const bool canEstimateWind = ((finalInflightYawInit && dragFusionEnabled) || assume_zero_sideslip()) &&
                                 !onGround &&
                                 PV_AidingMode != AID_NONE;
```

I wonder if maybe it is just a mixup between "finalInflightYawInit" and "yawAlignComplete" which can be seen about 10 lines lower?  I don't really understand the difference between these two variables.

I've asked @priseborough his opinion.  Perhaps @tridge will have an opinion too

priseborough, 2024-07-24:
It should be calling recordYawResetsCompleted() when the EKF yaw is reset to the GPS yaw

I'll submit a PR.

priseborough, 2024-07-24:
Ref https://github.com/ArduPilot/ardupilot/pull/27636

rmackay9, 2024-07-25:
Fixed in master and we can include it in an upcoming beta, thanks @priseborough

closed 2024-07-25 by rmackay9