ArduPilot/ardupilot #26654: Copter: APM_Config.h changes lead to "duplicate definition" compile error

rmackay9, 2024-03-30:
In Copter-4.4 developers could change APM_Config.h to include/exclude features but with 4.5 (and higher) this results in a compile error.

How to reproduce:

- git checkout Copter-4.4
- ./Tools/gittools/submodule-sync.sh
- in APM_config.h modify the line below:
    ` //#define MODE_GUIDED_NOGPS_ENABLED   DISABLED      // disable guided/nogps mode support`
    to be:
    `#define MODE_GUIDED_NOGPS_ENABLED   ENABLED      // disable guided/nogps mode support`
- ./waf distclean
- ./waf configure --board MatekF405-TE
- ./waf copter

The above will build successfully.  Then repeat with 4.5 and note the following error

```
In file included from ../../ArduCopter/config.h:29,
                 from ../../ArduCopter/UserParameters.cpp:2:
../../ArduCopter/APM_Config.h:21: error: "MODE_GUIDED_NOGPS_ENABLED" redefined [-Werror]
   21 | #define MODE_GUIDED_NOGPS_ENABLED   DISABLED      // disable guided/nogps mode support
      | 
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
```

Similarly with Copter-4.5 I found that ADVANCED_FAILSAFE could not be enabled through APM_Config.h

If we want to change how users include/exclude features that may be OK but we should clarify how this is done on the dev wiki and also remove misleading entries in APM_Config.h.

rbeall, 2024-03-30:
Technically this is just a Warning right?   I often times turn down my GCC warning thresh hold on Mac because it's set to high.

Another possible quick fix is likely to use undef right before that as it's being defined somewhere else I'm guessing.

Poruchik111, 2024-03-31:
Let me explain what problem.
In Copter 4.4.4. I have GuidedNoGps mode and Sport mide "from the box". I can set them on disarmed copter via Flight Mode switch.
In Copter 4.5.0/4.6.0. Dev - Sport mode and GuidedNoGps are disabled in Config.h 
But if I set them "ENABLED" and compile apj- I have Sport mode ok,  but GuidedNoGps "No such mode 20" error.
I checked everything- I cant find what the problem is. Nothing changed to Disable this mode. Guided mode work.
I even changed "minimize common.inc"
I need this mode for special GPS-less RC failsafe function.

APM Config.h no worked total as Randy listed abow.
I even don't care about it and see no problem to remoove it and forget. We have duplicated Config.h where disable func work well.

peterbarker, 2024-04-01:
> Let me explain what problem. In Copter 4.4.4. I have GuidedNoGps mode and Sport modes "from the box". I can set them on disarmed copter via Flight Mode switch. In Copter 4.5.0/4.6.0. Dev - Sport mode and GuidedNoGps are disabled in Config.h I understend you'r logic- GuidedNoGps mode made for script which not possible on 1 mB FC. I need this mode for special GPS-less RC failsafe function.
> 
> If I set them "ENABLED" and compile apj- I have Sport mode ok, but GuidedNoGps "No such mode 20" error. I checked everything- I can't find what the problem is. Nothing changed to Disable this mode. Guided mode work. I even changed "minimize common.inc"
> 
> APM Config.h no worked total as Randy listed abow. I even don't care about it and see no problem to remoove it and forget. We have duplicated Config.h where disable func work well. Thanks.

Yep, I think we do need to remove the lines from APM_config.h.

I've created https://github.com/ArduPilot/ardupilot/pull/26657 which will allow you to turn things on/off on the command-line.

... but the simplest way to do it is something like this:
```
cat >/tmp/extra.hwdef <<EOF
undef MODE_GUIDED_NOGPS_ENABLED
define MODE_GUIDED_NOGPS_ENABLED 1
EOF

./waf configure --board=MY_ONEMB_BOARDNAME --extra-hwdef=/tmp/extra.hwdef
./waf copter
```

This is actually just how the custom build server does things.

rmackay9, 2024-04-01:
@peterbarker,

Thanks again for this.  I think the issue is that it is difficult for developers to know what the name of the definition is.  APM_Config. is essentially the same as extra.hwdef (shown above) but it lists out the commonly used definitions.

Can't we just treat APM_Config.h the same as a user created extra.hwdef?

peterbarker, 2024-04-01:
> Thanks again for this. I think the issue is that it is difficult for developers to know what the name of the definition is. APM_Config. is essentially the same as extra.hwdef (shown above) but it lists out the commonly used definitions.

Perhaps we could look at some sort of command-line utility to help find the right define?

We do have plans to allow all of these build options.py via waf.  Perhaps that would fix the problem? eg.

```
./waf configure --board=Pixhawk1-1M --enable-feature=MODE_GUIDED_NOGPS_ENABLED
```
or
```
./waf configure --board=Pixhawk1-1M --enable-feature=GuidedNoGPS
```
(using the "name" field from `build_options.py`)

and perhaps
```
./waf list-features
```
... which would textualize `build_options.py`
 
> Can't we just treat APM_Config.h the same as a user created extra.hwdef?

Sadly we can't, as these are all vehicle-specific defines.  Things coming in in `extra.hwdef` work across the entire codebase.

rmackay9, 2024-04-01:
Hi @peterbarker,

Txs again.  I think the --enable-feautre=MODE_GUIDED_NOGPS_ENABLED would be great.  If we could list multiple features (separate arguments of course) and do the tab completion thing then we could just delete APM_Config.h or perhaps replace its contents with comments explaining how to use the command line to do the same thing.

peterbarker, 2024-04-01:
> Txs again. I think the --enable-feautre=MODE_GUIDED_NOGPS_ENABLED would be great. If we could list multiple features (separate arguments of course) and do the tab completion thing then we could just [...]

OK, I'll try to push ahead with this.

Note that not everything in `APM_config.h` can or can be represented in `build_options.py` at the moment, but we can move towards that goal.

Poruchik111, 2024-04-01:
Thanks guys for help and attention. I'll try now to check and report about.

If you dont mind- I want reccomend some changes from my opinion.

- Is it possible to make some complately new way to Enable/Disable AP feauches ?
F.e. aka Custom Build Server? You just take a file with all flight modes, sensors, CAN support... and set them "1" or "0".
Then you send command like " waf -config AP" and then compile binary as usual.

- I posted in Discuss my vision about VTX tables. Lot of new VTX have own VTX tables, it's hard to support all of them, so Ithink best way is separate VTX table constructor wich write your table directly in to EEPROM f.e. or add this func into Config file. You better know. Just think about.

- in OSD ELRS RSSI max signal level I see is 98%. LQ=100 and at low signal they a same- 23/23 f.e.
So I changed AP_OSD/ Screen.cpp line 1420:

 const uint8_t rssiv = ap_rssi->read_receiver_rssi() * 100; to
 const uint8_t rssiv = ap_rssi->read_receiver_rssi() * **102**;

Now I have max RSSI 100.

-OSD  Draw Callsighn- now used SD Card, Better to add Callsighn into Config.

- AP_Videotx.cpp- line 503 I add in
void AP_VideoTX::announce_vtx_settings() const
    **if(!hal.util->get_soft_armed())**
    GCS_SEND_TEXT(MAV_SEVERITY_INFO, "VTX: %s%d %dMHz, PWR: %dmW",

So now when you disarmed and change VTX Power by RC channel- you see full line "Channel L1 2362 MHz 1000mW blablabla"
When you armed and fly you not see it. Only VTX Power in OSD if you Enable it. Very helpfull. Only neccessary info in OSD.

-Defines.h have lot of unused definitions. Like other files. I think time to clean them all.

-I use Unicore UM980/ 982 GNSS module. Beginning 4.0.0. it supported as NMEA (GPS Type 5) and as Unicore (Type 24-25).
But no info how to config this type of GNSS. I spend lot of time and only when looked into Driver code I found how to configure it correct as NMEA. But I still can't config it as Unicore. (???)

APP: https://gnss.store/blog/post/unicore-uprecise-application.html
There is link and instruction, But! Agrica not used in NMEA AP driver! So right config which I use is:

In command line factory reset command-  **freset**    Wait.
in Config menu:
**GGA 0.2
RMC 0.2
VTG 0.2**
Check "Mode Rover", Check "Antenna feed on"
In command line:
**mask 20** (to mask poor signal og long distance Sats)
**config com1 115200** (not neccassary)
**saveconfig**
No neccassary connect GNSS RX pin to FC TX. 

-Std GNSS module send bad HDOP as 99,99 what calculated as HDOP 168.0.  Some expencive GNSS like Mozaic and Asterix have problem- when we have bad GPS and HDOP is high, they send nothing. This calculated by driver as HDOP = 0.0 and drone gone crazy.
So I added into AP_NavEKF3_VehicleStatus.cpp in line 169 check for HDOP=0:
 bool hdopFail = ((gps.get_hdop(preferred_gps) > 250) || gps.get_hdop(preferred_gps) == 0)) && (frontend->_gpsCheck & MASK_GPS_HDOP); 
Now copter work well.

Poruchik111, 2024-04-01:
This

cat >/tmp/extra.hwdef <<EOF
undef MODE_GUIDED_NOGPS_ENABLED
define MODE_GUIDED_NOGPS_ENABLED 1
EOF

made my day!!!
Really thanks!
my own logic work well now. Tomorrow will check whole SW.

Last question for today- after I remoove in WSL-2 local  repo and clone repo again- SITL not work well untill I load SIM SW in Mission Planner.
**param set sim_rc_fail** not worked and simulation have some bugs.

And now I have new problem- Telemetry log: mav.tlog ERROR: Not enough free disk space for logfile
I can't find these logs to delette them...
I use WSL connection in Visual Studio.
Thanks again

rmackay9, 2024-04-02:
@Poruchik111,

Maybe I'm misunderstanding but in SITL, the logs will normally appear in a "logs" directory in the directory where you started the simulator from.  So for example I start SITL in /ardupilot/ArduCopter then normally they will appear in a logs directory.

peterbarker, 2025-05-11:
OK, so I think tridge implemented what was desired in here - the ability to turn things on/off on the command-line.

That combined with creating little extra.hwdef files (and using the custom build server!) is sufficient to consider this issue closed, I think.

Maybe a little PR to say, "don't edit this file (APM_config.h), use `./waf configure --enable-XYZZY` instead?

rmackay9, 2025-05-12:
Hi @peterbarker,

Thanks again for checking back on all these various issues.

Yes, if users can enable and disable all the various items in the APM_Config.h then we can consider this closed.  Re how to inform the user, yes, an update to APM_Config.h and/or an update to the developer wiki to tell all users how to enable/disable features would be great.