ArduPilot/ardupilot #26679: guided mode

yuxuanwu1234567, 2024-04-03:
My unmanned car has two different motors. I adjusted them in the servo of the mission planner to be able to use remote control for linear walking, but in guided mode, they cannot move to the specified point and will still rotate in place. Is this a motor issue？

yuxuanwu1234567, 2024-04-03:
@Randy Mackay

yuxuanwu1234567, 2024-04-03:
@rmackay9

closed 2024-04-03 by rmackay9

yuxuanwu1234567, 2024-04-03:
This is my log. Thank you for your reply.@rmackay9
[2024-04-03 15-40-44.zip](https://github.com/ArduPilot/ardupilot/files/14849636/2024-04-03.15-40-44.zip)

rmackay9, 2024-04-04:
Hi @yuxuanwu1234567,

Let's continue the discussion on the support forums. https://discuss.ardupilot.org/t/yuxuanwu1234567s-skid-steering-rover/115746

Also I really need a .bin log.  .tlogs are a bit useless sadly.