ArduPilot/ardupilot #26681: GPS: `GPS_TYPE` missing values like `SBF-DualAntenna` in 4.5.0

flyingthingsintothings, 2024-04-03:
## Bug report

**Issue details**

When using Mission Planner `1.3.81 build 1.3.8741.25556` to connect to a device (Pixhawk 6X) running ArduPilot 4.5.0 (`Copter-4.5.0`), some values do not show up for the `GPS_TYPE` parameter. For example, the `SBF-DualAntenna` value is missing, and should normally appear at the end of the list with values.

**Version**

ArduPilot: `Copter-4.5.0`
Mission Planner: `1.3.81 build 1.3.8741.25556`

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Copter

**Hardware type**

Pixhawk 6X

**Logs**

\/

**Additional Information**

This was reported to @rmackay9 on Discord. This issue is also mentioned in https://github.com/ArduPilot/ardupilot/issues/26103

IamPete1, 2024-04-03:
I think https://github.com/ArduPilot/ardupilot/pull/26497 will have nuked all the meta data for the old param format. We really do  need to get param meta data versioning for mission planner.

flyingthingsintothings, 2024-04-03:
FYI @SeptentrioGNSS

flyingthingsintothings, 2024-04-03:
> We really do need to get param meta data versioning for mission planner.

Currently it's quite hard to know what version of the ArduPilot codebase is being used to get/generate the parameters, so that could be a helpful addition that makes the process used to get the parameters into Mission Planner a bit more understandable, especially for newer ArduPilot contributors like myself.

I know that I've seen the value `SBF-DualAntenna` displayed in Mission Planner at some point, so this issue got introduced close to the `4.5.0` release. I have seen the issue for a short while, but thought that it was my parameter files being broken as I had changed them manually for a PR. I recently removed all Mission Planner files (search all directories containing "mission planner" and remove them) and reinstalled Mission Planner to be sure, and the issue is still there. Others have also confirmed the issue.

rmackay9, 2024-04-03:
As @IamPete1 says, I think this really is a Mission Planner shortcoming but in any case I think we will need to put the GPS_TYPE2, etc parameter descriptions back into master to resolve this.

FYI @meee1

IamPete1, 2024-04-04:
> As @IamPete1 says, I think this really is a Mission Planner shortcoming but in any case I think we will need to put the GPS_TYPE2, etc parameter descriptions back into master to resolve this.
> 

As far as I know we don't host param docs on the server for anything other than master, so a GCS has only one option.

amilcarlucas, 2024-04-04:
@rmackay9 I do not think so! A better solution is to create version specific apm.pdef.xml files and put them on formware.ardupilot.org on the version specific folders.

That way every GCS would be able to get the correct version for the firmware they are seeing.

Long term to have outdated parameters in the apm.pdef.xml files will just confuse stuff!

flyingthingsintothings, 2024-06-11:
Closing as this seems to be resolved in `Copter-4.5.3`.

closed 2024-06-11 by flyingthingsintothings