ArduPilot/ardupilot #26690: NAVIO2：Provide a source without preprocessor for feature inclusion/exclusion.

muramura, 2024-04-04:
## Feature request

**Is your feature request related to a problem? Please describe.**

Many built-in preprocessors for functions are described, making the process difficult to understand.

**Describe the solution you'd like**

Provide a source that does not have a built-in function preprocessor.

**Describe alternatives you've considered**

Provide source for Raspi.

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

None

muramura, 2024-04-06:
[] I am currently removing the AP_XXXX preprocessor.

rmackay9, 2024-04-08:
I agree that the preprocessor lines are ugly and make the code harder to read.  They are important though because they allow AP to turn on/off features so that the application can fit onto all the boards we support and makes the custom build server possible.

Sadly I don't see an alternative to what we're doing.

muramura, 2024-04-08:
@rmackay9 san
Thanks for the comment.
ArduPilot also supports FC with no 2M byte limit.
I would remove the preprocessor for FC without the 2M byte limit.
GIT can merge changes in favor of new code.
This feature works well.

WORKING:
https://github.com/ArduPilot/ardupilot/compare/master...muramura:AP_RASPI

peterbarker, 2024-07-31:
We are not going to maintain two branches, one with `#if`s and one without.  It would be completely unmaintainable.

We *need* the preprocessor defines or most of our stm32 boards would stop working.

closed 2024-07-31 by peterbarker