ArduPilot/ardupilot #26697: AP_HAL_ChibiOS: add support for FMUv3 or CUAV Pixhack V3x with 3 IMUs

rmackay9, 2024-04-05:
We have a [request here](https://discuss.ardupilot.org/t/ardupilot-custom-firmware-builder-fmuv3-missing-necessary-for-cuav-pixhack-v3x/115823) to add Custom Build Server support for the FMUv3 so that the CUAV Pixhack V3x firmware supports 3 IMUs.

While [custom build server issues should be logged over here](https://github.com/ArduPilot/CustomBuild/issues) I've raised it here because I think we would not want to enable FMUv3 because we prefer that all boards are named properly.  So instead we should add a new build target for the CUAV Pixhack V3x (perhaps simply inheriting from FMUv3).

Harald69, 2024-04-05:
I think to add CUAV Pixhack V3x as a named build would be excellent.

closed 2024-08-06 by tridge

Harald69, 2024-08-06:
Thanks a lot