ArduPilot/ardupilot #26704: CANFD data stream hangs up after couple of minutes.

ketanatzebu, 2024-04-06:
## Bug report

**Issue details**

I have a proximity sensor giving out measurements via CANFD. I tried using lua script to read and process the frame but the sensor hangs up after few minutes. I have confirmed that it’s not an issue with the sensor using the CANFD analyzer on which I get uninterrupted continuous readings.

Nom Bit Rate: 1 000 000
Data Bit Rate: 5 000 000

Sensor: Ti mmwave AWR6843AOP

**Version**
Copter 4.5

**Platform**
[  ] All
[  ] AntennaTracker
[ . ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Cube Orange Plus

**Logs**
------- Clock Config -------
CAN_CLK_FREQ: 80MHz
Std Timings: bitrate=1000000 presc=8
sjw=1 bs1=8 bs2=1 sample_point=90.00000%
FD Timings: bitrate=5000000 presc=1
sjw=4 bs1=11 bs2=4 sample_point=75.00000%
------- CAN Interface Stats -------
tx_requests: 0
tx_rejected: 0
tx_overflow: 0
tx_success: 0
tx_timedout: 0
tx_abort: 0
rx_received: 388
rx_overflow: 0
rx_errors: 0
num_busoff_err: 0
num_events: 388
ECR: FF00
fdf_rx: 388
fdf_tx_req: 0
fdf_tx: 0

ketanatzebu, 2024-04-08:
> How are you determining that it hangs? Just that the received count stops increasing? Does it always stop after the same count? It would be useful if you could also upload the script.

I am printing the output on the messages section of the Mission Planner. It stops showing output when it hangs, After pressing reset on the sensor, The messages start showing up again for next couple of minutes. The count varies every time.

Here is the script:

`-- This script is an example of reading from the CAN bus

-- local default params
local update_rate_ms    = 1

local param_num_lua_prx_backend = 15
local lua_prx_backend
local lua_prx_driver_found = false
-- Load CAN driver1. The first will attach to a protocol of 10, the 2nd to a protocol of 12
-- this allows the script to distinguish packets on two CAN interfaces
local driver1 = CAN:get_device(25)
local get_xy = true
type = 0
length = 0

function setup_prx_sensor()
   if not driver1 then
     gcs:send_text(0,"No scripting CAN interfaces found")
     return
   end
   local sensor_count = proximity:num_sensors() -- number of sensors connected
   gcs:send_text(0,string.format("found %d sensors", sensor_count))
   for j = 0, sensor_count -1 do
     local device = proximity:get_backend(j)
     if ((not lua_prx_driver_found) and  device and (device:type() == param_num_lua_prx_backend)) then
       -- this is a lua driver
       lua_prx_driver_found = true
       lua_prx_backend = device
     end
   end
   if not lua_prx_driver_found then
     -- We can't use this script if user hasn't setup a lua backend
     gcs:send_text(0, string.format("Configure Lua prx Sensor"))
     return
   end
 end

function show_frame(dnum, frame)
   gcs:send_text(0, string.format("CAN[%u] msg from %i", dnum, frame:id_signed()))
end

-- Calculate power of 2
local function pow2(exponent)
   return 2 ^ exponent
end

-- Function to convert a 32-bit unsigned integer to a floating-point number
-- Convert uint32 to float
function uint32_to_float(uint)
   -- Extracting sign bit, exponent bits, and mantissa bits
   local sign = (uint >> 31) == 0 and 1 or -1
   local exponent = (uint >> 23) & 0xFF
   local mantissa = uint & 0x7FFFFF

   -- Handling special cases
   if exponent == 0 and mantissa == 0 then
      return 0.0
   elseif exponent == 0 and mantissa ~= 0 then
      return sign * math.ldexp(mantissa / 0x800000, -126)
   elseif exponent == 0xFF and mantissa == 0 then
      return sign * math.huge
   elseif exponent == 0xFF and mantissa ~= 0 then
      return 0.0 / 0.0
   end

   -- Computing the exponent (bias subtracted from the exponent bits)
   exponent = exponent - 127

   -- Computing the mantissa (including the implicit leading bit)
   local fraction = 1.0 + mantissa / 0x800000

   -- Constructing the floating-point number
   return sign * fraction * (2 ^ exponent)
end

function extract_and_convert(frame, index)
   return uint32_to_float((frame:data(index + 3) << 24) | (frame:data(index + 2) << 16) | (frame:data(index + 1) << 8) |
   frame:data(index))
end

function update()
   -- see if we got any frames
   if not lua_prx_driver_found then
      setup_prx_sensor()
   end
  
   if driver1 then
      local frame = driver1:read_frame()
      if not frame then
         return
      end

      if frame:id_signed() == 0 then
         type = frame:data(3) * 256 ^ 3 + frame:data(2) * 256 ^ 2 + frame:data(1) * 256 + frame:data(0)
         length = frame:data(7) * 256 ^ 7 + frame:data(6) * 256 ^ 6 + frame:data(5) * 256 + frame:data(4)
      elseif frame:id_signed() == 209 then
         local index = 0
         while index < 64 do
            -- x = uint32_to_float((frame:data(3) << 24) | (frame:data(2) << 16) | (frame:data(1) << 8) | frame:data(0))
            -- y = uint32_to_float((frame:data(7) << 24) | (frame:data(6) << 16) | (frame:data(5) << 8) | frame:data(4))
            -- z = uint32_to_float((frame:data(11) << 24) | (frame:data(10) << 16) | (frame:data(9) << 8) | frame:data(8))
            -- velocity = uint32_to_float((frame:data(15) << 24) | (frame:data(14) << 16) | (frame:data(13) << 8) | frame:data(12))

            -- Extracting and convertinsg x, y, z, and velocity from the frame
            local x = extract_and_convert(frame, index)
            local y = extract_and_convert(frame, index + 4)
            local z = extract_and_convert(frame, index + 8)
            -- local velocity = extract_and_convert(frame, index + 12)

            index = index + 16
            -- gcs:send_text(0, string.format("Point %d: X=%.2.f, Y=%.2f, Z=%.2f", 0, x, y, z))
            if not x == (0.00 and y == 0.00 and z == 0.00) then
               gcs:send_text(0, string.format("Point %d: X=%.2f, Y=%.2f, Z=%.2f", 0, x, y, z))
               -- if not (x == 0.00 and y == 0.00 and z == 0.00) then
               local point = Vector3f()
               point:x(x)
               point:y(y)
               point:z(z)
               lua_prx_backend:handle_script_3d_msg(point, true)
                  
            end
         end
      end
   end
   -- return update, 2
end

-- wrapper around update(). This calls update() and if update faults
-- then an error is displayed, but the script is not stopped
function protected_wrapper()
   local success, err = pcall(update)
   if not success then
       gcs:send_text(0, "Internal Error: " .. err)
       -- when we fault we run the update function again after 1s, slowing it
       -- down a bit so we don't flood the console with errors
       return protected_wrapper, 1000
   end
   return protected_wrapper, 1
 end
 
 -- start running update loop
 return protected_wrapper()
`

IamPete1, 2024-04-09:
I ran luacheck on your script, it finds a few issues:
```
    scripts/CAN test.lua:2:7: unused variable update_rate_ms
    scripts/CAN test.lua:10:7: unused variable get_xy
    scripts/CAN test.lua:11:1: setting read-only global variable type
    scripts/CAN test.lua:41:16: unused function pow2
    scripts/CAN test.lua:58:21: accessing undefined field ldexp of global math
    scripts/CAN test.lua:94:10: setting read-only global variable type
```
It would be useful to double check the script is still running. Can you enable disarmed logging and set `SCR_DEBUG_OPTS` bit 3 (value 8) for "log runtime memory usage and execution time".

ketanatzebu, 2024-04-10:
```
-- This script is an example of reading from the CAN bus

-- local default params
local update_rate_ms    = 1

local param_num_lua_prx_backend = 15
local lua_prx_backend
local lua_prx_driver_found = false
-- Load CAN driver1. The first will attach to a protocol of 10, the 2nd to a protocol of 12
-- this allows the script to distinguish packets on two CAN interfaces
local driver1 = CAN:get_device(25)

local function setup_prx_sensor()
   if not driver1 then
     gcs:send_text(0,"No scripting CAN interfaces found")
     return
   end
   local sensor_count = proximity:num_sensors() -- number of sensors connected
   gcs:send_text(0,string.format("found %d sensors", sensor_count))
   for j = 0, sensor_count -1 do
     local device = proximity:get_backend(j)
     if ((not lua_prx_driver_found) and  device and (device:type() == param_num_lua_prx_backend)) then
       -- this is a lua driver
       lua_prx_driver_found = true
       lua_prx_backend = device
     end
   end
   if not lua_prx_driver_found then
     -- We can't use this script if user hasn't setup a lua backend
     gcs:send_text(0, string.format("Configure Lua prx Sensor"))
     return
   end
 end

-- Function to convert a 32-bit unsigned integer to a floating-point number
-- Convert uint32 to float
local function uint32_to_float(uint)
   -- Extracting sign bit, exponent bits, and mantissa bits
   local sign = (uint >> 31) == 0 and 1 or -1
   local exponent = (uint >> 23) & 0xFF
   local mantissa = uint & 0x7FFFFF

   -- Handling special cases
   if exponent == 0 and mantissa == 0 then
      return 0.0
   elseif exponent == 0 and mantissa ~= 0 then
      return sign * math.ldexp(mantissa / 0x800000, -126)
   elseif exponent == 0xFF and mantissa == 0 then
      return sign * math.huge
   elseif exponent == 0xFF and mantissa ~= 0 then
      return 0.0 / 0.0
   end

   -- Computing the exponent (bias subtracted from the exponent bits)
   exponent = exponent - 127

   -- Computing the mantissa (including the implicit leading bit)
   local fraction = 1.0 + mantissa / 0x800000

   -- Constructing the floating-point number
   return sign * fraction * (2 ^ exponent)
end

local function extract_and_convert(frame, index)
   return uint32_to_float((frame:data(index + 3) << 24) | (frame:data(index + 2) << 16) | (frame:data(index + 1) << 8) |
   frame:data(index))
end

local function update()
   -- see if we got any frames
   if not lua_prx_driver_found then
      setup_prx_sensor()
   end
   gcs:send_text(0, "Script Still Running")
   if driver1 then
      local frame = driver1:read_frame()
      if not frame then
         return
      end

      if frame:id_signed() == 0 then
         local type = frame:data(3) * 256 ^ 3 + frame:data(2) * 256 ^ 2 + frame:data(1) * 256 + frame:data(0)
         local length = frame:data(7) * 256 ^ 7 + frame:data(6) * 256 ^ 6 + frame:data(5) * 256 + frame:data(4)

         gcs:send_text(0, string.format("TLV Type: %d  |  TLV Length: %d", type, length))
      elseif frame:id_signed() == 209 then
         local index = 0
         while index < 64 do
            -- Extracting and convertinsg x, y, z, and velocity from the frame
            local x = extract_and_convert(frame, index)
            local y = extract_and_convert(frame, index + 4)
            local z = extract_and_convert(frame, index + 8)
            -- local velocity = extract_and_convert(frame, index + 12)

            index = index + 16
            -- gcs:send_text(0, string.format("Point %d: X=%.2.f, Y=%.2f, Z=%.2f", 0, x, y, z))
            if not (x == 0.00 and y == 0.00 and z == 0.00) then
               gcs:send_text(0, string.format("Point %d: X=%.2f, Y=%.2f, Z=%.2f", 0, x, y, z))
               -- if not (x == 0.00 and y == 0.00 and z == 0.00) then
               local point = Vector3f()
               point:x(x)
               point:y(y)
               point:z(z)
               lua_prx_backend:handle_script_3d_msg(point, true)
            end
         end
      end
   end
   -- return update, 2
end

-- wrapper around update(). This calls update() and if update faults
-- then an error is displayed, but the script is not stopped
local function protected_wrapper()
   local success, err = pcall(update)
   if not success then
       gcs:send_text(0, "Internal Error: " .. err)
       -- when we fault we run the update function again after 1s, slowing it
       -- down a bit so we don't flood the console with errors
       return protected_wrapper, 1000
   end
   return protected_wrapper, update_rate_ms
 end
 -- start running update loop
 return protected_wrapper()

```

I have updated the parameters as requested and the script as well.
I added an extra message which confirms that the script is not hanging up even after the sensor reading stops.

I have also noticed that when the sensor reading stops, the sensor also hangs ( Which is not the case with classic CAN messages, I am getting continuous stream. Tested it for more than 1hr )  but this only happens when it is connected to flight controller. As I mentioned, I've checked the sensor with the CANFD analyzer and there is not interruption in the stream.

Note: The sensor is externally powered.

rmackay9, 2024-04-11:
Hi @ketanatzebu,

I'm not too familiar with this area of the code but I guess you tried modifying the CAN_P1_FDBITRATE parameters and it didn't help?

rmackay9, 2024-04-11:
.. anyway, I guess this is resolved so I'm going to go ahead and close this issue if that's OK. I've ping'd other developers to get their input on whether we should support 5Mbps or not.

ketanatzebu, 2024-04-15:
> Hi @ketanatzebu,
> 
> I'm not too familiar with this area of the code but I guess you tried modifying the CAN_P1_FDBITRATE parameters and it didn't help?

Yes, I configured my sensor for 1Mbps and changed the param CAN_P1_FDBITRATE to 1M.

vasurk, 2024-06-06:
Hi @ketanatzebu ,

How did you configure the mmwave sensor and what lua scirpt you have added could you please inform?

Thanks