ArduPilot/ardupilot #26706: Unable to install firmware speedybeef4v4 into SpeedyBee F405 V4

adimkov, 2024-04-06:
## Bug report

Unable to install firmware [speedybeef4v4](https://firmware.ardupilot.org/Copter/stable/speedybeef4v4/) into SpeedyBee F405 V4 

I tried to install speedybeef4v4 into SpeedyBee F405 V4 with the tool 'STM32CubeProgrammer', but it constantly failed with the error `Error: Data mismatch found at address  0x08000003 (byte = 0x00 instead of 0x20)` on validation. 
Here is the log of the programming that cause the issue (I think)
```
  12:51:50 : sector 0000 does not exist
  12:51:50 : sector 0001 does not exist
  12:51:50 : sector 0002 does not exist
  12:51:50 : sector 0003 does not exist
  12:51:50 : sector 0004 does not exist
  12:51:50 : sector 0005 does not exist
  12:51:50 : sector 0006 does not exist
  12:51:50 : sector 0007 does not exist
  12:51:50 : sector 0008 does not exist
  12:51:50 : sector 0009 does not exist
  12:51:50 : sector 0010 does not exist
  12:51:50 : sector 0011 does not exist

```
The Flight Controller is brand new; however, I'm not sure if it's original because it was bought on AliExpress.

Note: Betta flight works perfectly fine.

**Version**
[stable-4.5.0](https://firmware.ardupilot.org/Copter/stable-4.5.0)

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
SpeedyBee F405 V4

**Logs**
Programming:
```
12:51:48 : Read File: /Users/anton/Downloads/arducopter_with_bl.hex
  12:51:48 : Number of segments: 1
  12:51:48 : segment[0]: address= 0x8000000, size= 0xE7D30
  12:51:50 : Memory Programming ...
  12:51:50 : Opening and parsing file: arducopter_with_bl.hex
  12:51:50 :   File          : arducopter_with_bl.hex
  12:51:50 :   Size          : 927.30 KB 
  12:51:50 :   Address       : 0x08000000 
  12:51:50 : Erasing memory corresponding to segment 0:
  12:51:50 : Erasing internal memory sectors [0 11]
  12:51:50 : sector 0000 does not exist
  12:51:50 : sector 0001 does not exist
  12:51:50 : sector 0002 does not exist
  12:51:50 : sector 0003 does not exist
  12:51:50 : sector 0004 does not exist
  12:51:50 : sector 0005 does not exist
  12:51:50 : sector 0006 does not exist
  12:51:50 : sector 0007 does not exist
  12:51:50 : sector 0008 does not exist
  12:51:50 : sector 0009 does not exist
  12:51:50 : sector 0010 does not exist
  12:51:50 : sector 0011 does not exist
  12:51:50 : Download in Progress:
  12:52:06 : File download complete
  12:52:07 : Time elapsed during download operation: 00:00:16.218
```
Verify: 
```
12:53:00 : Error: Data mismatch found at address  0x08000003 (byte = 0x00 instead of 0x20)
```

adimkov, 2024-04-07:
Hello!

I have found a workaround for this. The utility dfu-util can install a bootloader and, through MissionPlanner, firmware.

```dfu-util -a 0 --dfuse-address 0x08000000 -D speedybeef4v3_bl.bin```

The issue can be closed. 

Thank you, 
Anton

closed 2024-04-07 by IamPete1

valposv, 2024-05-18:
Downgrading from STM32CubeProgrammer 2.16.0 to 2.15.0 helps as well