ArduPilot/ardupilot #26731: Build Fails When Configured For Debug

davidbitton, 2024-04-09:
## Bug report

**Issue details**

Build fails at the very end while linking. This error occurs -- at least -- when building a MATEKF450-CAN and a MATEKH743 flight controller with `--debug` included during configuration. The error does not appear when building for SITL.

**Version**
master

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
all

**Hardware type**
maybe all

**Logs**
```bash
arm-none-eabi/bin/ld:common.ld:202 cannot move location counter backwards (from 00000000081c0960 to 00000000081c0000)
```

IamPete1, 2024-04-09:
"cannot move location counter backwards"  means it ran out of flash.

Without debug there is 166672 bytes free. If I define out OSD then it builds with debug with 7216 bytes free. Adding debug symbols adds a lot size to the build.

closed 2024-04-09 by IamPete1

davidbitton, 2024-04-09:
@IamPete1 commenting out all of the OSD bits the hwdef.dat allowed me to build an H743. The MatekF405-CAN is so packed already, i'll just use the H743 if I want to track crashdumps.