ArduPilot/ardupilot #26757: Request support for yge esc telemetry

212049, 2024-04-11:


212049, 2024-04-11:
Hello, I want to get the telemetry support of yge esc, yge now opens the open yge protocol, and rotorfilght also supports this protocol, could you please port it?

IamPete1, 2024-04-12:
These are the Rotorflight PRs: https://github.com/rotorflight/rotorflight-firmware/pulls?q=OpenYGE+

From a quick look through the Rotorflight implementation it certainly looks like something we could add support for. (it would be quite easy with scripting too)

@bob01 Looks like you have been heading the effort for Rotorflight, is the protocol documented anywhere?

bob01, 2024-04-12:
Hi Peter -

Yes, I was fortunate to reach the right ears at YGE and used the
opportunity extended to co-author OpenYGE w/ YGE.

It's just recently complete so no documentation beyond comments and working
notes.

The implementation in RF RC1 was based on a draft which has been
superseded by the final spec which we refer to as v3.
I'd be happy to provide pointers to a more current reference implementation
(which will make it's way into RF at some point after RF2) and some light
docs to get you going.
I was once part of the ArduPilot project - will be happy to help.

- Rob