ArduPilot/ardupilot #26785: Build error compiling 4.5.1 Copter with F405 (4.4.x) target

lida2003, 2024-04-12:
a) F405 (4.4.x) target build ok
b) pull  4.5.1 code, build error

I'm NOT sure what config parameter caused this.

```
ChibiOS: Done!

[ 95/998] Compiling libraries/AP_Airspeed/Airspeed_Calibration.cpp
[ 96/998] Compiling libraries/AP_Airspeed/AP_Airspeed_analog.cpp
[ 97/998] Compiling libraries/AP_Airspeed/AP_Airspeed_SDP3X.cpp
[ 98/998] Compiling libraries/AP_Airspeed/AP_Airspeed_Health.cpp
[ 99/998] Compiling libraries/AP_Airspeed/AP_Airspeed_MS4525.cpp
In file included from ../../libraries/AP_GPS/AP_GPS.h:31,
                 from ../../libraries/AP_Airspeed/AP_Airspeed_Health.cpp:8:
../../libraries/GCS_MAVLink/GCS_MAVLink.h:40:10: fatal error: include/mavlink/v2.0/all/version.h: No such file or directory
   40 | #include "include/mavlink/v2.0/all/version.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Waf: Leaving directory `/home/daniel/SnapDragonfly/ardupilot/build/Aocoda-RC-F405V3'
Build failed
 -> task in 'objs/AP_Airspeed' failed (exit status 1):
        {task 125452233812432: cxx AP_Airspeed_Health.cpp -> AP_Airspeed_Health.cpp.0.o}
 (run with -v to display more information)
```

rmackay9, 2024-04-12:
Support issues should really go into [the support forums](/) so I'll close this.  Sorry to be strict but if we're not then our to-do list turns into another support forum.

I'm pretty sure that 4.5.1 is building correctly so perhaps you need to update submodules.  The best way to do that I think is to enter:

- cd ardupilot
- ./Tools/gittools/submodule-sync.sh

closed 2024-04-12 by rmackay9

lida2003, 2024-04-12:
It has something to do with `define AP_TRAMP_ENABLED 1` and `include ../include/minimize_fpv_osd.inc` or `include ../include/minimize_features.inc`

Just for memory

andyp1per, 2024-04-12:
You probably just need to run "waf clean"

lida2003, 2024-04-12:
> You probably just need to run "waf clean"

@rmackay9 @andyp1per Thanks. 

It's a little bit strange here, not code issue.

lida2003, 2024-05-14:
> > You probably just need to run "waf clean"
> 
> @rmackay9 @andyp1per Thanks.
> 
> It's a little bit strange here, not code issue.

It has something to do with bashrc script. It's good to go when I disable this line. 

I'm NOT dive into this, but I didn't find `bash` dir in ardupilot.  Well I'm NOT sure about what's this  all about.

```
$ tail -n 1 ~/.bashrc
#source /home/daniel/SnapDragonfly/ardupilot/Tools/completion/completion.bash

$ cat /home/daniel/SnapDragonfly/ardupilot/Tools/completion/completion.bash
#!/usr/bin/env bash

_AP_COMPLETION_DIR=$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" > /dev/null && pwd)
source "$_AP_COMPLETION_DIR/bash/_waf"
source "$_AP_COMPLETION_DIR/bash/_sim_vehicle"
source "$_AP_COMPLETION_DIR/bash/_ap_bin"
source "$_AP_COMPLETION_DIR/bash/_ap_autotest"
```