ArduPilot/ardupilot #26800: AP_VisualOdom.h：a bug in get_delay_ms(), return type error

luweiagi, 2024-04-15:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_DAL/AP_DAL_VisualOdom.h#L21
[image]
 
This function is used at:
https://github.com/ArduPilot/ardupilot/blob/5a21d0cb8a859b293eea087fb8c9be150c0ca910/libraries/AP_NavEKF3/AP_NavEKF3_core.cpp#L83
[image]

We can see that its return should be uint16_t or uint32_t, not bool.

@peterbarker

rmackay9, 2024-04-15:
That does indeed look like a bug.  Thanks @luweiagi!

tridge, 2024-04-16:
@luweiagi great catch!

peterbarker, 2024-04-16:
Good find.  4-year-old bug!

rmackay9, 2024-04-17:
Fixed in master and this will go out with 4.5.2-beta1.  @luweiagi thanks again!

closed 2024-04-17 by rmackay9