ArduPilot/ardupilot #26801: SITL: X-Plane 11 communication not working

mkatliar, 2024-04-15:
## Bug report

**Issue details**

When ArduPlane is started with `--model xplane` option, there is no reaction from X-Plane. Steps to reproduce:

1. Configure X-Plane: 
[image]
2. Start a flight in X-Plane
3. Run ArduPlane: 
    ```
    $ ~/ardupilot/build/sitl/bin/arduplane --model xplane
    Waiting for XPlane data on UDP port 49001 and sending to port 49000
    Loaded 20 DRefs from @ROMFS/models/xplane_plane.json
    Starting sketch 'ArduPlane'
    Starting SITL input
    Using Irlock at port : 9005
    bind port 5760 for SERIAL0
    SERIAL0 on TCP port 5760
    Waiting for connection ....
    ```
4. Connect to AuduPlane from a GCS. ArduPlane says:
    ```
    Connection on serial port 5760
    bind port 5762 for SERIAL1
    SERIAL1 on TCP port 5762
    bind port 5763 for SERIAL2
    SERIAL2 on TCP port 5763
    Home: -35.363262 149.165237 alt=584.000000m hdg=353.000000
    validate_structures:514: Validating structures
    XPlane version 115501
    ```
5. Try to takeoff the airplane from GCS, or anything else -- there is no reaction from X-Plane.

X-Plane does send UDP data to port 49001:
```
$ sudo tcpdump -i any udp port 49001 -vv -X
...
12:00:01.408295 lo    In  IP (tos 0x0, ttl 64, id 10941, offset 0, flags [DF], proto UDP (17), length 69)
    localhost.49001 > localhost.49001: [bad udp cksum 0xfe44 -> 0xfce4!] UDP, length 41
        0x0000:  4500 0045 2abd 4000 4011 11e9 7f00 0001  E..E*.@.@.......
        0x0010:  7f00 0001 bf69 bf69 0031 fe44 4441 5441  .....i.i.1.DDATA
        0x0020:  2a01 0000 006b c412 43e5 3c7d 40e5 3c7d  *....k..C.<}@.<}
        0x0030:  4000 0000 0000 c079 c440 02a0 4181 0440  @......y.@..A..@
        0x0040:  4195 7459 3e                             A.tY>
12:00:01.464732 lo    In  IP (tos 0x0, ttl 64, id 10944, offset 0, flags [DF], proto UDP (17), length 69)
    localhost.49001 > localhost.49001: [bad udp cksum 0xfe44 -> 0xf0f7!] UDP, length 41
        0x0000:  4500 0045 2ac0 4000 4011 11e6 7f00 0001  E..E*.@.@.......
        0x0010:  7f00 0001 bf69 bf69 0031 fe44 4441 5441  .....i.i.1.DDATA
        0x0020:  2a01 0000 0027 d212 431b 3a80 401b 3a80  *....'..C.:.@.:.
        0x0030:  4000 0000 0000 c079 c448 02a0 418f 0440  @......y.H..A..@
        0x0040:  413e 7859 3e                             A>xY>
```
However nothing comes out of ArduPlane on port 49000:
```
sudo tcpdump -i any udp port 49000 -vv -X
<... silence ...>
```

I attached the debugger to `arduplane` and set a breakpoint in `XPlane::receive_data()`. I can see that the execution reaches this line: https://github.com/ArduPilot/ardupilot/blob/5a21d0cb8a859b293eea087fb8c9be150c0ca910/libraries/SITL/SIM_XPlane.cpp#L357, so `RREF` are received. But this line https://github.com/ArduPilot/ardupilot/blob/5a21d0cb8a859b293eea087fb8c9be150c0ca910/libraries/SITL/SIM_XPlane.cpp#L362 is never reached, so `DATA` packets never get processed -- although `tcpdump` shows them being sent to port 49001. 

I have no idea why this happens. 

**Version**
Built from master branch, commit hash 5a21d0cb8a859b293eea087fb8c9be150c0ca910

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X-Plane

**Hardware type**
SITL

**Logs**
See issue details

mkatliar, 2024-04-21:
Same with X-Plane 10

tggo, 2025-01-05:
yep, some problem.
you find any solution?

tggo, 2025-01-05:
[image]
[image]
[image]

ok, if i comment ```send_drefs()``` it start communication normal

tggo, 2025-01-05:
ok. i found problem in X-Plane 12
need change "Port we send from (legacy)" to any not 49001, because it conflict with ArduPilot simulator  - [image]