ArduPilot/ardupilot #26811: ROS2 / DDS install issues (discussion1, discussion2) -- caused by user's installed tools falling out of date.

Ryanf55, 2024-04-16:


seantfish, 2025-05-03:
Given microxrceddsgen is not yet available as a binary, I have a potential solution for making compilation easier:
- Add microxrceddsgen to the .repos file
- Use colcon-gradle to build microxrceddsgen 
- Create an ament_cmake shim package that adds the microxrceddsgen bash script to the bin directory, such that the script is available on PATH automatically

With this new workflow, all users need to do is:
- Install the correct version of Java
- Install colcon-gradle with pip

seantfish, 2025-05-26:
Curious, are you on ROS Humble or ROS Jazzy? I am currently running Jazzy.

Ryanf55, 2025-05-26:
> Another note, I am having trouble building the upstream eProsima version but the ArduPilot fork does build fine (on Jazzy).
> 
> Edit: ArduPilot fork of Micro-XRCE-DDS-Gen also builds on my Humble dev container
> 
> Edit 2: I replicated the error you got when building from inside Micro-XRCE-DDS-Gen [@Ryanf55](https://github.com/Ryanf55) - what error do you get when building from the workspace?

1) I'm using humble, we are slowly adding support for Jazzy. PR's welcome for issues you find.
2) If I build from a workspace it works, just not inside the repo. seems like a bug from the tool. It doesn't seem to add the executable to PATH though.
```
(venv-ardupilot) ryan@B650-970:~/Dev/ardu_ws/src/test_env$ colcon build
Starting >>> Micro-XRCE-DDS-Gen
Finished <<< Micro-XRCE-DDS-Gen [10.5s]                    

Summary: 1 package finished [10.6s]
(venv-ardupilot) ryan@B650-970:~/Dev/ardu_ws/src/test_env$ . install/setup.bash
w(venv-ardupilot) ryan@B650-970:~/Dev/ardu_ws/src/test_env$ which microxrceddsgen 
/home/ryan/Dev/ros2_ws/src/Micro-XRCE-DDS-Gen/scripts/microxrceddsgen
```

Do you know how we coudl automate that? It's still using the repo from my ~/.bashrc.

seantfish, 2025-05-27:
So `colcon-gradle` doesn't install to PATH, but it does install to CLASSPATH (https://github.com/colcon/colcon-gradle/blob/main/colcon_gradle/task/gradle/build.py). 

I've created a shim package that makes use of the CLASSPATH rather than a reference to a built jar.
Try this fork: https://github.com/ASDL-Robotics/ardupilot/tree/add-shim-dep
It fails building but I think that is a bug from master, not caused by gradle.

seantfish, 2025-05-27:
Also, `pexpect` is not declared as a dependency in the package.xml files currently. I'm guessing this is a package that is common on some Ubuntu distributions but doesn't ship with my container environment.

Jai-GAY, 2025-06-05:
I am running ROS 2 Humble on Ubuntu 22.04 Jammy.
I encountered an issue with [$./gradlew assemble build](https://discuss.ardupilot.org/t/colcon-build-packages-up-to-ardupilot-dds-tests-failed/115720/9) for Micro-XRCE-DDS-Gen.